solettaproject/soletta #18: Complete build options for Pthread API

elfring, 2015-06-18:
Would you like to add the configuration script "[AX_PTHREAD](http://www.gnu.org/software/autoconf-archive/ax_pthread.html "How to build C programs using POSIX threads?")" to [your build specification](https://github.com/solettaproject/soletta/blob/3f5cffa1a1a507ddc93f0ff17df03e363ce786cf/configure.ac#L122 "Configuration checks")?

lpereira, 2015-06-18:
The build system is going to be changed shortly, so there's no need to change that at this moment.

barbieri, 2015-06-18:
being more specific, we're looking into moving to kbuild.

as it may take some  time to be fully tested and merged, I'd accept the patch to use AX_PTHREAD as it shouldn't hurt.

mbelluzzo, 2015-06-30:
Since kbuild/kconfig is in, I'm closing this issue.

closed 2015-06-30 by mbelluzzo