solettaproject/soletta #19: Completion of error handling

elfring, 2015-06-18:
Would you like to add more error handling for return values from functions like the following?
- [pthread_mutex_unlock](http://pubs.opengroup.org/onlinepubs/9699919799/functions/pthread_mutex_unlock.html "Unlock a mutex.") ⇒ [sol_worker_thread_unlock](https://github.com/solettaproject/soletta/blob/936ff840d70bfd282afb7aa28caa9043626853ea/src/shared/sol-worker-thread-pthread.c#L77)
- [strdup](http://pubs.opengroup.org/onlinepubs/9699919799/functions/strdup.html "Duplicate a string.") ⇒ [runner_new](https://github.com/solettaproject/soletta/blob/936ff840d70bfd282afb7aa28caa9043626853ea/src/bin/sol-fbp-runner/runner.c#L109)

barbieri, 2015-06-18:
could be, these are pretty unlikely to fail in their conditions but we can get a patch to handle those :-)

the strdup() we usually check, I guess this one wasn't being checked as the runner_new() is basically only used on Linux and it does over-allocations, in such case pretty unlike to fail on strdup(). But it's a good practice to check it.

the mutex unlock error from manpage seem like neve happening in our usage:
```
EPERM  The mutex type is PTHREAD_MUTEX_ERRORCHECK or
              PTHREAD_MUTEX_RECURSIVE, or the mutex is a robust mutex, and the
              current thread does not own the mutex.
```

barbieri, 2015-08-27:
will close this issue since no replies. According to my comment, this is not required.

closed 2015-08-27 by barbieri

elfring, 2015-08-27:
I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Unchecked return values") a bit more.

Are you interested to apply [aspect-oriented software development](http://aspectc.org/ "AspectC++")?
How do you think about to encapsulate error detection and corresponding exception handling as [a reusable aspect](http://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in your software?

barbieri, 2015-08-27:
@elfring, as I said in  my previous comment, the possible error `EPERM` can't happen as we're not setting such flags. We can add the error checking in the case someone changes the mutex flags at creation time, but I'd like to avoid such checks "just because".

elfring, 2015-08-27:
I would prefer that strict error detection will become less optional in important software.
Is [the importance of software robustness](https://en.wikipedia.org/w/index.php?title=Robustness_%28computer_science%29 "Making software more robust?") increasing also for your framework?