solettaproject/soletta #79: Out-of-source build is broken

thiagomacieira, 2015-06-25:
When you try to configure out-of-source, you get:

 $ $srcdir/autogen.sh
 python3: can't open file 'autogen.sh/data/oic/oicgen.py': [Errno 2] No such file or directory

dorileo, 2015-06-25:
This oic generator being called out of the build system itself (namely being called from autogen.sh) seems not the right thing, the new build system does that - if we want to fix it before kconfig gets integrated we should move the oic generator to inside the build system itself.

barbieri, 2015-06-25:
Autogen.sh is to be executed in tree. Not just for oicgen, but autoconf as well. Some sources are generated and should be distributed in the "make dist". As well as Makefile.in

What you can execute out of tree is configure.

dorileo, 2015-06-25:
Yep, I know, what I mean is that the oic generator should be called from Makefile (like we for for all the other generators) and not Autogen.sh (the way it's today).

glima, 2015-07-02:
Can we close this now that kbuild got in?

closed 2015-07-03 by mbelluzzo