solettaproject/soletta #80: Failure to build with kernel headers 3.16 or older

thiagomacieira, 2015-06-25:
Error was:

```
src/shared/sol-network-linux.c: In function ‘sol_network_link_up’:
src/shared/sol-network-linux.c:479:20: error: ‘IN6_ADDR_GEN_MODE_EUI64’ undeclared (first use in this function)
     uint8_t mode = IN6_ADDR_GEN_MODE_EUI64;
                    ^
src/shared/sol-network-linux.c:479:20: note: each undeclared identifier is reported only once for each function it appears in
src/shared/sol-network-linux.c:515:65: error: ‘IFLA_INET6_ADDR_GEN_MODE’ undeclared (first use in this function)
     ADD_RTATTR(addr_gen_container, RTA_LENGTH(sizeof(uint8_t)), IFLA_INET6_ADDR_GEN_MODE);
                                                                 ^
src/shared/sol-network-linux.c:508:27: note: in definition of macro ‘ADD_RTATTR’
         _attr->rta_type = _type;                                              \
                           ^
```

OpenSUSE 13.2 comes with kernel 3.16. This is the latest version.

dorileo, 2015-06-25:
We have seen this issue with yocto builds, and have maintained a patch for that. Maybe not the proper fix but this is what I have in hands:

---
 src/shared/sol-util.h | 16 ++++++++++++++++
 1 file changed, 16 insertions(+)

diff --git a/src/shared/sol-util.h b/src/shared/sol-util.h
index 13f20ac..340a131 100644
--- a/src/shared/sol-util.h
+++ b/src/shared/sol-util.h
@@ -7,6 +7,22 @@
 #include <string.h>
 #include <time.h>
 
+#define IFLA_INET6_UNSPEC 0
+#define IFLA_INET6_FLAGS 1
+#define IFLA_INET6_CONF 2
+#define IFLA_INET6_STATS 3
+#define IFLA_INET6_MCAST 4
+#define IFLA_INET6_CACHEINFO 5
+#define IFLA_INET6_ICMP6STATS 6
+#define IFLA_INET6_TOKEN 7
+#define IFLA_INET6_ADDR_GEN_MODE 8
+#define __IFLA_INET6_MAX 9
+
+#define IFLA_INET6_MAX (__IFLA_INET6_MAX - 1)
+
+#define IN6_ADDR_GEN_MODE_EUI64 0
+#define IN6_ADDR_GEN_MODE_NONE 1
+
 #define streq(a, b) (strcmp((a), (b)) == 0)
 #define streqn(a, b, n) (strncmp((a), (b), (n)) == 0)
 
-- 
2.4.1

ceolin, 2015-06-25:
sol-missing.h is the place for these defines, i'm going to check a proper fix to this.

Thanks for pointing it.

dorileo, 2015-06-25:
Sounds like a better fix, thanks.

ceolin, 2015-06-25:
The fix  is in the pullreq https://github.com/solettaproject/soletta/pull/87

ceolin, 2015-06-26:
Pullreq already integrated, i'm closing this issue.

closed 2015-06-26 by ceolin