solettaproject/soletta #89: Can't include both linux/i2c.h and linux/i2c-dev.h if i2c-dev is provided by i2c-tools

thiagomacieira, 2015-06-25:
Whenever linux/i2c-dev.h is provided by i2c-tools, it is incompatible with linux/i2c.h.

Errors:
```
In file included from src/shared/sol-i2c-linux.c:39:0:
/usr/include/linux/i2c-dev.h:37:8: error: redefinition of ‘struct i2c_msg’
 struct i2c_msg {
        ^
In file included from src/shared/sol-i2c-linux.c:38:0:
/usr/include/linux/i2c.h:68:8: note: originally defined here
 struct i2c_msg {
        ^
In file included from src/shared/sol-i2c-linux.c:39:0:
/usr/include/linux/i2c-dev.h:89:7: error: redefinition of ‘union i2c_smbus_data’
 union i2c_smbus_data {
       ^
In file included from src/shared/sol-i2c-linux.c:38:0:
/usr/include/linux/i2c.h:128:7: note: originally defined here
 union i2c_smbus_data {
       ^
```

See https://build.opensuse.org/package/view_file/openSUSE:13.2/linux-glibc-devel/linux-glibc-devel.spec line 102

glima, 2015-07-13:
Since I'm starting vacations real soon and I could not reproduce it on time, @ceolin is going to chase the issue instead.

thiagomacieira, 2015-07-13:
Install OpenSUSE to reproduce. Or at the very least ensure that /usr/linux/linux/i2c-dev.h comes from the i2c-tools package and is unmodified from upstream.

This is most likely an OpenSUSE/i2c-tools bug. I'm asking for a workaround.

ceolin, 2015-07-16:
@thiagomacieira the following commit does the workaround. I've tested installing the header from i2c-tools manually and worked. Could you confirm it to me close this issue ?

https://github.com/solettaproject/soletta/commit/427f47fb2c7698d06438a243600d5c0c5aa5758d

thiagomacieira, 2015-07-16:
If you're getting that warning, I guess it fixes the issue, because I'm getting the warning too.

Why do you need to include linux/i2c.h? I didn't need to here.

ceolin, 2015-07-16:
linux/i2c-dev.h from i2c-tools contains the symbols from linux/i2c.h. But in the common case where i2c-dev.h comes from kernel it's necessary to include linux/i2c.h

thiagomacieira, 2015-07-16:
Understood.

closed 2015-07-16 by ceolin