solettaproject/soletta #189: irange-buffer.fbp test failing when running with valgrind

cmarcelo, 2015-07-02:
```
$ make check-fbp-valgrind
FAILED running irange-buffer.fbp: wrong exit code, got 1 but expected 0
Output in /tmp/check-fbp.Wk82Xt

ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'test_median2' failed

tools/build/Makefile.targets:17: recipe for target 'check-fbp-valgrind' failed
make: *** [check-fbp-valgrind] Error 1
```

cmarcelo, 2015-08-03:
@tcanabrava I thought this one disappeared but is still around. Could you take a look at it later?

tcanabrava, 2015-08-04:
sure.

glima, 2015-08-07:
The timeout test is the one on trouble under Valgrind.

tcanabrava, 2015-08-07:
@cmarcelo , 

While running make check-fbp-valgrind I got tons of errors:

FAILED running /data/soletta/src/test-fbp/boolean-or.fbp: wrong exit code, got 1 but expected 0
Output:
==12168== 77,402 (368 direct, 77,034 indirect) bytes in 1 blocks are definitely lost in loss record 674 of 674
==12168==    at 0x4C2C080: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==12168==    by 0x56A726F: ??? (in /usr/lib/libcurl.so.4.3.0)
==12168==    by 0x5391ADB: sol_http_client_init (in /data/soletta/build/soletta_sysroot/usr/lib/libsoletta.so.0.0.1)
==12168==    by 0x5385AC9: sol_comms_init (in /data/soletta/build/soletta_sysroot/usr/lib/libsoletta.so.0.0.1)
==12168==    by 0x5379017: sol_init (in /data/soletta/build/soletta_sysroot/usr/lib/libsoletta.so.0.0.1)
==12168==    by 0x402424: main (in /data/soletta/build/soletta_sysroot/usr/bin/sol-fbp-runner)
==12168== 

The error seems to be that the curl stuff is closed on an idler, but the idler wasn't triggered untill the end of the program, and so a leak warning was issued.

glima, 2015-08-13:
I think this one can be closed now.

bdilly, 2015-08-13:
Curl issue was solved.
I've run check-fbp-valgrind and only one test was skipped because timer expired at wrong time thanks to valgrind.
So I think we're good here.

closed 2015-08-13 by bdilly