solettaproject/soletta #287: `sol-flow-builtopts.h` included from two different places break pragma once

cmarcelo, 2015-07-11:
During the compilation this file ends up being included from two different places. I've found curious that during compilation we include headers from sysroot instead of staging / src.

```
      CC   build/stage/lib/flow/sol-flow-resolver.o
In file included from ./src/lib/flow/sol-flow-resolver.c:37:0:
./src/lib/flow/sol-flow-buildopts.h:44:40: warning: redundant redeclaration of ‘sol_flow_resolver_conffile’ [-Wredundant-decls]
 extern const struct sol_flow_resolver *sol_flow_resolver_conffile;
                                        ^
In file included from ./build/soletta_sysroot/usr/include/soletta/sol-flow.h:228:0,
                 from ./build/soletta_sysroot/usr/include/soletta/sol-flow-resolver.h:37,
                 from ./src/lib/flow/sol-flow-resolver.c:35:
./build/soletta_sysroot/usr/include/soletta/sol-flow-buildopts.h:44:40: note: previous declaration of ‘sol_flow_resolver_conffile’ was here
 extern const struct sol_flow_resolver *sol_flow_resolver_conffile;
                                        ^
      CC   build/stage/lib/flow/sol-flow-builder.o
```

cmarcelo, 2015-07-11:
@dorileo any idea what's going on?

cmarcelo, 2015-07-13:
In IRC, @ibriano suggested that we put all generated files in `build/stage` and during compilation, only include files from either there or the source directory, i.e. NOT include files from `sysroot`. This would fix the issue above.

dorileo, 2015-07-13:
I have a patch to generate *-buildopt.h directly to build_sysroot instead of src, it fixes the issue... I plan to send lt later today or in the worst scenario tomorrow.

cmarcelo, 2015-07-28:
Ping

dorileo, 2015-07-29:
Ok, I'm bringing it back to the top of my todo list... thanks for pinging.

dorileo, 2015-07-31:
fixed with commit 59f3de1d4f9e6694bda0a0e33e2b9d6e7d8a7c49

closed 2015-07-31 by dorileo

cmarcelo, 2015-08-10:
> In IRC, @ibriano suggested that we put all generated files in build/stage and during compilation, only > include files from either there or the source directory, i.e. NOT include files from sysroot. This would > fix the issue above.

@dorileo @ibriano  Should we create a separate issue for this?

ibriano, 2015-08-10:
I can get to that now, I was about to tackle some dependency problems I found when not using the default config, so I'm still neck deep in the build system.