solettaproject/soletta #377: sol-fbp-runner has misleading error message and doesn't accept unexpected node metadata

davidmaxwaterman, 2015-07-27:
##### Confusing error message.
FBP  files that have nodes containing unexpected metadata cause the Soletta fbp runner to fail with a misleading error message. For example, nodes contain metadata defining a label :

```
node9(constant/string:label=node9) OUT -> IN node10(console:label=node10)
```

Causes the runner to output :

```
...:1:1 Couldn't resolve type name 'constant/string'
```

which makes me think there is something wrong with the type name, but that's not the problem at all. The actual problem is the runner doesn't accept the 'label=' metadata.

Please consider changing the error message to something more representative of the actual error, eg:

```
Error parsing node: node9 - unexpected metadata 'label'
```

##### Unexpected metadata
FBP files containing unexpected metadata also cause the fbp runner to fail. Please consider changing this behaviour so that users can embed information that is useful for purposes that haven't been considered. Specifically, allowing `label="node9"` should be allowed, but a better solution would be to simply ignore metadata that is unexpected, else you'd be forcing developers of other tools to maintain input and output processors to make it so the file is acceptable as input to Soletta.

Thanks.

barbieri, 2015-07-27:
Hi Max,

Thanks for your email, I've assigned @ricardotk to look at it since he worked in the parser and error messages, should be able to deliver the fix soon.

As for your proposal regarding the unexpected metadata, it won't be implemented. We will not accept unexpected metadata in the properties, otherwise it will defeat the purpose of check and correctness.

If you need to store extra information, do as we're doing for the tests of FBP: use a comment in the line below or after it. We're using this to make a single file where the FBP is created and the output is tested.

As for your particular example, "label", this should be the actual node name (as you did, label=node9 for the node9). Why do you want to duplicate this?

Moreover, I think it's perfectly fine to save your information as a different format such as JSON, just offer the way to get the FBP from it so it can be executed in the sol-fbp-runner. If you opt to do it, we can also create a Python script and keep that in our repository (we can do this part of the task, you do the JS version embedded in DAVE, we do the Python inside Soletta).

glima, 2015-08-07:
I guess @cmarcelo already solved it upstream, right?

cmarcelo, 2015-08-13:
The error messages were improved in 24a13c75ab1df08fd97f038957a766d9167c574b. In the reported example

    node9(constant/string:label=node9) OUT -> IN node10(console:label=node10)

the error now is

    test.fbp:1:23 Unknown option name 'label' of node 'node9'

which resolves the issue.

About the suggestion of being liberal on these values, the downside would be that typos and small errors could be undetected. There was a discussion with Max about where to hold metadata editor (reason for this suggestion) and it was agreed that it should be kept in a different file.

closed 2015-08-13 by cmarcelo