solettaproject/soletta #401: The workaround to set the GPIO pins for Linux Kernel version is >=  3.18 doesn't work.

dborquez, 2015-07-29:
The workaround for Linux Kernel version is >=  3.18 proposed in /soletta/src/samples/flow/minnow-calamari/README doesn't set the GPIO pins properly.

Path to the fbp files: /soletta/src/samples/flow/minnow-calamari/

fpb files tested:

1. calamari-rgb-led.fbp
2. calamari-lever.fbp
3. calamari-led.fbp
4. calamari-buttons-rgb-led.fbp
5. calamari-7seg.fbp

Errors returned:

1. calamari-rgb-led.fbp:42:28 Couldn't resolve type name 'RGBLed'
2. calamari-lever.fbp:35:1 Couldn't resolve type name 'SegmentsLever'
3. calamari-led.fbp:36:1 Couldn't resolve type name 'LedLever'
4. calamari-buttons-rgb-led.fbp:38:1 Couldn't resolve type name 'Button1'
5. calamari-7seg.fbp:38:26 Couldn't resolve type name 'SevenSegments'

tcanabrava, 2015-07-29:
How did you launched the sol-fbp-runner? 
did you exported SOL_FLOW_MODULE_RESOLVER_CONFFILE to be the sol-flow.json ?

dborquez, 2015-07-29:
yes I did it first with;

     export SOL_FLOW_MODULE_RESOLVER_CONFFILE=sol-flow-new.json

and after that by launch the fbp-runner:

      sol-fbp-runner <fbp file name>

tcanabrava, 2015-08-07:
I'v commited a fix for that a day or two ago, can you test again?

dborquez, 2015-08-07:
Test done. The patch have fixed the issue. Thank you.

closed 2015-08-07 by dborquez