solettaproject/soletta #408: soletta building from source fails

dborquez, 2015-07-30:
- steps to reproduce the issue:

1. git clone https://github.com/solettaproject/soletta.git
2. make alldefconfig
3. make

- the step (3) fails returning the following output:

' ./src/lib/common/sol-worker-thread-impl-glib.c:54:45: error: ‘__ATOMIC_SEQ_CST’ undeclared (first use in this function) '

- complete output here:  http://pastebin.intel.com/view/2eb2a875

tcanabrava, 2015-07-31:
try with a newer GCC - __ATOMIC was introduced with gcc 4.7

anselmolsm, 2015-07-31:
My bet is that soletta should require gcc>=4.7.

anselmolsm, 2015-07-31:
Also, it should require python>=3.4.
I remember there was an issue with versions older than that (I can't remember the issue itself).

dborquez, 2015-07-31:
It could be good to add the Installation Required Packages to build Soletta from source.

dborquez, 2015-07-31:
- I have updated the gcc and python packages to:
gcc 4.9.2
Python 3.4

- the gcc update solved the missing  ‘__ATOMIC_SEQ_CST’ declaration.
- A new issue appears in compilation:

' ./src/lib/flow/sol-flow-js.c:36:21: fatal error: duktape.h: No such file or directory
 #include "duktape.h" '
compilation terminated.

- The console output: http://pastebin.intel.com/view/6ca1d586

tcanabrava, 2015-07-31:
dukttape is a library that's cloned inside the soletta repository via a git submodule, it seems that it failed downloading it.
can you do a git clean -xdf and then make alldefconfig again?

dborquez, 2015-07-31:
I did the git clean command that you suggested but I got the same compilation output:

./src/lib/flow/sol-flow-js.c:36:21: fatal error: duktape.h: No such file or directory
 #include "duktape.h"

anselmolsm, 2015-07-31:
duktape is a javascript vm that is distributed as a pair of .c and .h files, meant to be built as part of the project. Due to several reasons, Soletta is fetching this pair of files as a git submodule (in fact, it is not an actual submodule).

git clean is not enough, more steps are needed to overcome this invalid state of the tree, we're investigating this.

davidmaxwaterman, 2015-08-07:
I would like to add that I also had big trouble building Soletta, while Elliot didn't - the difference seems to be that I use Ubuntu and he uses Fedora. The issues I came across seemed to be missing Python modules, which I found relatively simply using 'apt-cache search' and installed using 'apt-get install'. It would be nice if these dependencies were obtained automatically, or at least mentioned in the README.

barbieri, 2015-08-07:
+1 to mention in the README.md
-1 to install automatically as that's hard given how different distros handle packages. 

OTOH, should we add more fields to dependencies.json and print more useful information to the user when a test fail? Like a description that says why such test is being done ("description": "Checks for built in compiler support of integer overflow"), if it fails what to say ("on_failure": "using fallback overflow checks -- may not be as fast,  GCC 4.8 supports it").

The dependencies resolver should also have a way to print the test that failed. Maybe linked with "make V=1" to pass a --verbose. For pkg-config it would print the exact command line and execution envvars, with all output from pkg-config. For ccode it should also print the full test source and the compiler errors, as well as command line and envvars.

This is what autoconf's config.log was. We may generate such file as well if we wish.

More work for @dorileo

anselmolsm, 2015-08-13:
We're populating this [wiki page](https://github.com/solettaproject/soletta/wiki/Build-requirements) with the build requirements. Let's add to README.md a link to this page.

barbieri, 2015-08-27:
ping. Is this still pending?

anselmolsm, 2015-09-13:
* The build requirements wiki is being populated ( @dorileo++ for using tables).
 * The README section [Build from source](https://github.com/solettaproject/soletta#building-from-source) points to the wiki page mentioned above.
 * The duktape submodule black magic was replaced with actual git submodule, so we shouldn't face problems related to our script.
 * The dep-resolver results go to config.log

So, all issues pointed out here were covered. Closing, tks.

closed 2015-09-13 by anselmolsm