solettaproject/soletta #415: make failure /udev/udev.c:154:17: error: void value not ignored as it ought to be

dborquez, 2015-07-31:
-Plattform: Linux kernel 3.13.0-30-generic #55~precise1-Ubuntu SMP Fri Jul 4 21:52:00 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
-gcc 4.9.2
-make 3.81-8.1
-Python 3.4
-git 2.5.0

Steps to reproduce this issue:
1. git clone https://github.com/solettaproject/soletta.git 
2. make alldefconfig
3. make

complete output: http://pastebin.com/MMEacGXH

dorileo, 2015-08-03:
you're using an older libudev version.

dorileo, 2015-08-03:
The dependency-resolver should check for an specific version - since we support that. I'm checking what the proper version we support and we'll add that to dep-resolver configuration file.

glima, 2015-08-13:
I think it's ok now, isn't it?

anselmolsm, 2015-08-13:
@glima, the check for libudev is not there yet.
IIRC, to avoid such error due to API changes, libudev should be at least the one part of systemd 219.

glima, 2015-08-13:
I said that because at least the build on galileo's sdk does not show up the udev node anymore.

anselmolsm, 2015-08-13:
In my previous comment, should be "the check for libudev **version** is not there yet."

anselmolsm, 2015-08-18:
Not a P1 IMO, this happens in old systems and only blocks the build of a single node type: udev.
#556 added a check for minimum libudev version that will avoid the wrong dependency detection (and disable node type udev when necessary).

glima, 2015-08-18:
Definitely not a P1.

dborquez, 2015-08-18:
Please provide with the correct priority classification. Thank you.

anselmolsm, 2015-08-18:
Set P2, but considering if it should be a P3...

closed 2015-08-18 by anselmolsm