solettaproject/soletta #424: sol-fbp-generator fails to generate a .c file

dborquez, 2015-08-03:
In order to test the SLV feature to export fbp files, I tried to convert a fbp file to .c source code with the following command line:

'/sol-fbp-generator2 -j /usr/share/soletta/flow/descriptions/builtins.json -j /usr/share/soletta/flow/descriptions/calamari.json  SLV_Examples/slv_created_calamari_led_lever.fbp slv_calamari-lever.c'

- Output (error) returned by the previous command:

SLV_Examples/slv_created_calamari_led_lever.fbp:1:40 Expected ',' after key-pair meta information. e.g. '(nodetype:key1:val2,keyN:valN)'
Segmentation fault

- below is the content of the file: 'slv_created_calamari_led_lever.fbp'  :

node2(calamari/lever:poll_interval=max:"INT32_MAX"|min:"INT32_MIN"|step:1|val:0,range=max:1023|min:0|step:1|val:0,bus=max:"INT32_MAX"|min:"INT32_MIN"|step:1|val:0,chip_select=max:"INT32_MAX"|min:"INT32_MIN"|step:100|val:0) OUT -> INTENSITY node3(calamari/led:period=max:"INT32_MAX"|min:"INT32_MIN"|step:1|val:10000,range=max:10000|min:0|step:5|val:0,address=1)

Thank you!

barbieri, 2015-08-04:
I cannot try now, but the INT32_MAX and INT32_MIN shouldn't have quotes as they are constants. @cmarcelo may know more about it.

ricardotk, 2015-08-04:
@dborquez Like @barbieri said, I tested here and it worked removing the `"` for INT32_MAX and INT32_MIN values.

dborquez, 2015-08-04:
@barbieri and @barbieri, that's correct, with the changes before mentioned it works. I'll proceed now to close this issue and redirect this to SLV team.

closed 2015-08-04 by dborquez