solettaproject/soletta #463: warning in make output

davidmaxwaterman, 2015-08-07:
Following the README to build Soletta, I get a warning that sticks out like a sore thumb in the otherwise nice-and-clean output - but not indication if it is a problem or what to do about it :

```
$ make alldefconfig
...
$ make
...
./src/modules/flow/string/string.c:40:2: warning: #warning "You're building the string nodes module without i18n support -- some nodes will only act properly on pure ASCII input, not the intended utf-8 for Soletta" [-Wcpp]
 #warning "You're building the string nodes module without i18n support -- some nodes will only act properly on pure ASCII input, not the intended utf-8 for Soletta"
...
```

barbieri, 2015-08-07:
Good catch. Se should find a way to let users see important missing dependencies instead of silently disabling features that depend on it.

And after that we need to allow users forcing the features to be disabled. Like an "ACK, I know I miss that dependency" for missing dependencies. This would also work for systems that have a feature but it shouldn't be installed. I guess @glima had to remove icu to test the fallback path and that's not good.

@dorileo, would you come with a recommended way to do it and apply for icu, setlocale and newlocale?

glima, 2015-08-07:
Yeah, ideally this warning should be in the config menu. @dorileo, will you help on that? I can put in it my TODO if yours is already too big.

dorileo, 2015-08-11:
I'm looking at it...

dorileo, 2015-08-11:
I've talked to @glima and we found out that we really want to warn the user about the lacking of i18n support and so - since it considerably changes the overall module behavior. But we came up with a solution to better display this kind of warnings, I agree the #waning solution doesn't seem right and elegant, but we'll introduce something like ```obj-{mod}-warning``` to the build system, so we can properly display these messages to the user. I'm working on the solution.

davidmaxwaterman, 2015-08-11:
from a user point of view, I'd be interested to know how to fix the problem...

glima, 2015-08-11:
Install ICU development packages.

dorileo, 2015-08-11:
hi @glima 

I'll guess what @davidmaxwaterman  is mostly concerned about the message to be displayed, in that matter we can change the warning message to inform the user to install ICU dev packages...

glima, 2015-08-11:
Okay, fair. I'll add that.

barbieri, 2015-08-11:
@dorileo this warning shouldn't exist if the user disabled the support for some module. It's like the old autotools:
 * if want_XXX && !have_XXX: fail or warn;
 * if want_XXX && have_XXX: success;
 * if !want_XXX && (!have_XXX || have_XXX): success, no warnings.

Then the code and Makefiles will use "want_XXX" not "have_XXX" tests.

We should add the "have_XXX" as Kconfig always-visible items, then users will know they are missing some dependency and they can't enable a certain feature, such as "--- Use ICU", then go to its help and see "depends on HAVE_ICU".

dorileo, 2015-08-11:
absolutely...

dorileo, 2015-08-11:
The pull request #501 adds the basic infra for the warnings, we still need to adjust the configuration entries to comply with the "Use ICU" thing @barbieri  requested.

barbieri, 2015-08-27:
ping

I guess this is done in master.

glima, 2015-08-27:
Yes, closing.

closed 2015-08-27 by glima