solettaproject/soletta #487: build: there should be a way to config a debug build

cmarcelo, 2015-08-11:
Currently we have to always pass CFLAGS="-g" for every `make` call. If the flag is forgotten, the build might get a mix of objects compiled with different flags.

There should be an easy way (config option maybe?) to declare a build is a debug one, and don't have to worry about passing the right CFLAGS.

cmarcelo, 2015-08-11:
@dorileo @ibriano makes sense to you?

ibriano, 2015-08-11:
On Tue, 11 Aug 2015 06:38:46 -0700, Caio Marcelo de Oliveira Filho wrote:
> @dorileo @ibriano makes sense to you?
> 

Yes

barbieri, 2015-08-11:
yes, we should have a build options section in Kconfig, it should allow us to enable debug, sanitizer, lto, etc...

dorileo, 2015-08-11:
we've just integrated a patch regarding sanitizer, this may be chosen with kconfig (currently we've 2 sanitizer types there others - if required - may be added.

For flto we also have patch for that #482..

We could add an -g option, thinking about that now, we could also have an option for -O as well.

barbieri, 2015-08-27:
ping

I guess this is done in master.

barbieri, 2015-09-30:
ping 2... i guess this was done already

ibriano, 2015-09-30:
On Wed, 30 Sep 2015 11:46:58 -0700, Gustavo Sverzut Barbieri wrote:
> ping 2... i guess this was done already

There's a compiler options with release/debug choices plus setting of
arbitrary cflags and ldflags, so I guess that means yes.

closed 2015-09-30 by barbieri