solettaproject/soletta #525: PWM on the galileo gen 2 is not working

glima, 2015-08-13:
WRN:sol-pwm ./src/lib/io/sol-pwm-linux.c:214 _pwm_config() pwm #0,5: could not change polarity

is where it stops on the grove-led-accumulator.fbp sample.

Also, it seems that building for that board with networking enabled, seems to be crashing on the sol_init() -- I guess we should check the deps, if curl is being properly detected, etc.

dborquez, 2015-08-15:
Hello,

Could you please assign the priority of this bug by insert at the beginning of the title a P[X] where X =1,2,3 or 4, i.e. 1 is the highest priority.
Thank you

Thank you

tcanabrava, 2015-08-15:
what?

glima, 2015-08-17:
> Hello,
> 
> Could you please assign the priority of this bug by insert at the beginning of the title a P[X] where X =1,2,3 or 4, i.e. 1 is the highest priority.
> Thank you

OK, David, I'll do that. Let's say P1 is fair for this one.

> 
> 
> Thank you
> 
> ---
> https://github.com/solettaproject/soletta/issues/525#issuecomment-131323767

mbelluzzo, 2015-09-18:
I was able to have pwm 1 3 and 5 working on galileo with pull req #730

glima, 2015-09-18:
Nice! If you could run our galileo samples again it would be nice to assert they are now fine.

barbieri, 2015-09-30:
ping

mbelluzzo, 2015-09-30:
pwn is working on galileo, dunno about the sample, but the bug is related to the pwm, so I would close it

glima, 2015-09-30:
@dborquez , would you please test the galileo samples again (involving PWM), so that we can close this issue? @mbelluzzo says the root cause of the error should be now fixed. Thanks a lot :)

dborquez, 2015-10-01:
It was tested with the groove kit buzzer example, and it works fine.

anselmolsm, 2015-10-01:
QA tested and reported this as fixed. Closing, tks.

closed 2015-10-01 by anselmolsm