solettaproject/soletta #536: Convert sol-util functions to use sol-buffer

barbieri, 2015-08-14:
Functions like `sol_util_load_file_raw()` and `sol_util_fill_buffer()`should be using `sol_buffer` instead of own buffer manipulation functions.

This also have nice side-effects like allowing `sol_util_load_file_string()` getting the `sol_util_load_file_raw()` buffer and being able to ensure the trailing `\0` without the need of a `realloc()``

Also report on functions that would help this task, I already envision `sol_buffer_trim()` that calls `sol_buffer_resize(buf, buf->used)` that could be used at the string version before it returns. Maybe `sol_buffer_steal(buf)` to take `buf->data` and reset `buf->used = 0; buf->capacity = 0;`?

edersondisouza, 2015-08-18:
@barbieri 

    ...that could be used at the string version before it returns

So, `sol_util_load_file_string()` should start returning a `sol_buffer`? Won't it make its usage cumbersome?

barbieri, 2015-08-18:
I was thinking more in the lines of `sol_util_load_file_raw()` returning a buffer and `sol_util_load_file_string()` calling `sol_util_load_file_raw()`, forcing a trailing `\0`, calling `sol_buffer_resize(buf, buf->used)` and stealing the buffer's data, returning it.

Maybe these new calls should be introduced to help make it clear:
```c
int sol_buffer_trim(buf) {
    return sol_buffer_resize(buf, buf->used);
}

void *sol_buffer_steal(buf, size_t *size) {
    void *r = buf->data;
    if (*size) *size = buf->used;
    buf->data = NULL;
    buf->used = 0;
    buf->capacity = 0;
    return r;
}
```

edersondisouza, 2015-08-18:
Ok. Got it.

edersondisouza, 2015-08-24:
@barbieri does #589 close this issue? Is there anything missing?

barbieri, 2015-08-24:
yes.

closed 2015-08-24 by barbieri