solettaproject/soletta #543: tests contained in /src/test-fbp and src/samples/flow/unix-socket failed in Galileo Gen2 Plattform

dborquez, 2015-08-14:
Error messages like the following:

'Couldn't resolve type 'test/result' of node 'test_compare_red_byte' '

are returned on the execution of the fbp tests located in soletta/src/test-fbp.

The complete list of error messages is listed in the link below: 

http://pastebin.intel.com/view/46e6ddf4  are executed in a Galileoboard Gen2.

Thank you.

anselmolsm, 2015-08-14:
Please, provide the sha1.
Reopen with the proper info.

closed 2015-08-14 by anselmolsm

dborquez, 2015-08-14:
This is the sha1: ece20c7fde9c310f83b14048701b2c57619f2800

Error messages like the following:

'Couldn't resolve type 'test/result' of node 'test_compare_red_byte' '

are returned on the execution of the fbp tests located in soletta/src/test-fbp.

The complete list of error messages is listed in the link below: 

http://pastebin.intel.com/view/46e6ddf4 are executed in a Galileoboard Gen2.

Thank you.