solettaproject/soletta #551: Examples in src/flow/unix-socket failed to run in Edison board

dborquez, 2015-08-15:
Soletta sha1: 605718d8d24413ae6e4a200d0cc282a273627ec8

fbp tests contained in /src/samples/flow/unix-socket failed to run and returned the following output:

 WRN:sol-flow-unix-socket-impl ./src/modules/flow/unix-socket/unix-socket-impl.c:223 unix_socket_client_new() Could not connect: No such file or directory
WRN: ./src/modules/flow/unix-socket/unix-socket.c:364 byte_reader_open() mdata->un_socket== NULL
WRN:sol-flow ./src/lib/flow/sol-flow.c:137 sol_flow_node_init() failed to create node of type=0xb76a6620: Operation not permitted
WRN:sol-flow ./src/lib/flow/sol-flow-static.c:473 flow_node_open() failed to init node #0, type=0xb76a6620, opts=0x92f8c30: Operation not permitted
WRN:sol-flow ./src/lib/flow/sol-flow.c:137 sol_flow_node_init() failed to create node of type=0x92f8d40: Operation not permitted
ERR: ./src/bin/sol-fbp-runner/main.c:134 startup() Failed to run

The complete list of output error messages for is listed in the link below:

http://pastebin.com/f5vfCm57

Thank you.

mbelluzzo, 2015-09-21:
I got the same errors when I ran boolean-reader.fbp before boolean-writer.fbp. However, when running in the expected order, the sample worked just fine. (Beware that is easier to do this with 2 terminals on the board).

This "ordering" instructions could be found on boolean-reader.fbp file. I guess that this is also expected by all <thing>-reader/writer pair. @dborquez could you check if the problem is fixed by running the fbp's in the correct order?

dborquez, 2015-09-25:
You are correct mbelluzo, this was not a bug.

closed 2015-09-25 by dborquez