solettaproject/soletta #561: Reduce the amount of generated code outside build/stag

cmarcelo, 2015-08-17:
Our build generates a bunch of artifacts inside the source dir, some of these files shouldn't be there. The current state is

    $ make alldefconfig 2>&1 > /dev/null
    $ make -j9 2>&1 > /dev/null
    $ git clean -dfx
    Removing .config
    Removing .config-cache
    Removing Kconfig.gen
    Removing Makefile.gen
    Removing build/
    Removing data/scripts/sol-flow-node-type-find.py
    Removing data/scripts/sol-flow-node-type-gen.py
    Removing include/
    Removing src/lib/common/sol-pin-mux-builtins-gen.h
    Removing src/lib/flow/sol-flow-builtins-gen.h
    Removing src/modules/flow/oic/oic.c
    Removing src/modules/flow/oic/oic.c~
    Removing src/modules/flow/oic/oic.json
    Removing tools/kconfig/conf
    Removing tools/kconfig/conf.o
    Removing tools/kconfig/zconf.hash.c
    Removing tools/kconfig/zconf.lex.c
    Removing tools/kconfig/zconf.tab.c
    Removing tools/kconfig/zconf.tab.o

Except for the directory `build/` and the files `tools/kconfig/` (which I presume need to be there so we don't change kconfig itself), all the other files should be inside `build/` unless there's a compelling reason not to.

cmarcelo, 2015-08-17:
See also https://github.com/solettaproject/soletta/issues/287#issuecomment-129552930

dorileo, 2015-08-19:
currently the only files being generated in the source dir are:

```
src/lib/common/sol-pin-mux-builtins-gen.h
src/lib/flow/sol-flow-builtins-gen.h
src/modules/flow/oic/oic.json
./data/scripts/sol-flow-node-type-gen.py
./data/scripts/sol-flow-node-type-find.py
```

I'll see - in this list - if these files really need to be generated there.

dorileo, 2015-08-20:
fixed on d90e7371c3a49198816cf843c2d8adceee98e5d6

closed 2015-08-20 by dorileo