solettaproject/soletta #562: fbp-generator issues

zehortigoza, 2015-08-17:
- min/max option on conf file causing a invalid C file being generated

Using a conf to define the min/max option of a node cause a invalid C file being generated.

conf file:
```javascript
{
 "$schema": "http://solettaproject.github.io/soletta/schemas/config.schema",
 "nodetypes": [
  {
   "name": "BaseServo",
   "options": {
    "chip": 0,
    "duty_cycle_range": "min:1000|max:2000",
    "period": 20000,
    "pin": 0
   },
   "type": "servo-motor/controller"
  }
 ]
}
```
fbp file:
```javascript
base_servo(BaseServo)
```
invalid block generated:
```javascript
static const struct sol_flow_node_type *
create_0_root_type(void)
{
    static const struct sol_flow_node_type_servo_motor_controller_options opts0 =
        SOL_FLOW_NODE_TYPE_SERVO_MOTOR_CONTROLLER_OPTIONS_DEFAULTS(
            .chip = {
                .val = 0,
            },
            .duty_cycle_range = {
                ."min=1000,
                .max=2000",
            },
            .period = {
                .val = 20000,
            },
            .pin = {
                .val = 0,
            },
        );
...
```

- Duplicated include on generated file

Having 2 or mode nodes of the same type on the fbp cause duplicated includes on the generated file.

cmarcelo, 2015-08-18:
I'll take a look at this.

zehortigoza, 2015-08-18:
Thanks issues solved after #571 .

closed 2015-08-18 by zehortigoza