solettaproject/soletta #607: random: integer errors reported by -fsanitize=undefined

barbieri, 2015-08-21:
```
$ ./build/stage/samples/flow/misc/random-numbers 
...
src/modules/flow/random/random.c:90:42: runtime error: signed integer overflow: 666 * 1812433253 cannot be represented in type 'int [624]'
src/modules/flow/random/random.c:73:13: runtime error: left shift of 513147066 by 7 places cannot be represented in type 'int'
src/modules/flow/random/random.c:74:13: runtime error: left shift of 395728058 by 15 places cannot be represented in type 'int'
...
```

This is running on a 32bits machine, but shouldn't matter as the type is just "int" (not long/off_t/size_t).

assigned to @lpereira as he introduced this code as an alternative to missing `rand()` in libC.

lpereira, 2015-08-26:
Taking a look

lpereira, 2015-08-27:
Fixed in HEAD.

closed 2015-08-27 by lpereira

anselmolsm, 2015-08-27:
Fixed by #637