solettaproject/soletta #653: [P2] failed running soletta/src/test-fbp/bitwise-int.fbp in Minnowboard

dborquez, 2015-08-29:
soletta sha1: 69f00c94f6ea36807022bc1dd715429644709ee6
Plattform: Minnowboard

FAILED running /home/root/soletta/src/test-fbp/bitwise-int.fbp
FAILED running /home/root/soletta/src/test-fbp/javascript.fbp

output returned:

Output:
WRN:sol-flow ./src/lib/flow/sol-flow-resolver-conffile.c:189 _resolver_conffile_get_module() failed to get rootdir for module test
bitwise-byte.fbp:43:32 Couldn't resolve type 'test/result' of node 'test_byte_bitwise_and'

Thank you.

anselmolsm, 2015-08-31:
OK, in fact this is not a test error. The minnow image is being generated with Soletta without JS support (duktape is not being built).
I'm rebuilding the image, adding the step needed to include duktape.

cc: @brunobottazzini

brunobottazzini, 2015-09-14:
Issue retested with duktape support enabled and everything worked.

closed 2015-09-14 by brunobottazzini