solettaproject/soletta #663: [P2] string related tests contained in soletta/src/test-fbp/ failed to run in Galileo board G2.

dborquez, 2015-08-31:
soletta v1_beta2 sha1: 3baae03daa72e6c0587a17c7750a463c813b4e55

The following tests failed to run:

string-compare.fbp
string-lower-upper.fbp
string-concat.fbp
string-regexp-search.fbp
string-length.fbp

A complete list of returned errors is founded in the following link:

http://pastebin.com/hCVRyJZY

Thank you.

mbelluzzo, 2015-09-23:
string tests right now were maybe to stress the use of ICU. When compiling to galileo and edison, a priori you don't have ICU, so the default backend is used and thus those tests don't work. We will release a set of tests to exercise both backends soon.

mbelluzzo, 2015-09-30:
@dborquez can you confirm that this is solved? We now have ascii tests but the icu tests are still supposed to fail if you dont have icu.

dborquez, 2015-10-01:
I have tested the V1 Beta 6 (a1adfd1699fffae01e6ec0beda44bc7865923766) and it works.

closed 2015-10-01 by dborquez