solettaproject/soletta #665: sol-fbp-generator crashes

lgywata, 2015-09-01:
git bisect pointed to this commit: b37442e9391983ccbb5f125f679f87ccee39a8c5

$ sol-fbp-generator -j ./build/soletta_sysroot/usr/share/soletta/flow/descriptions/ ~/tmp/fbp/simple.fbp main.c
=================================================================
==466== ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60040000dfb8 at pc 0x7fcb1c8f6a89 bp 0x7fff438ff4c0 sp 0x7fff438ff4b8
READ of size 1 at 0x60040000dfb8 thread T0
    #0 0x7fcb1c8f6a88 (/usr/lib/libsoletta.so.0.0.1+0x8ea88)
0x60040000dfb8 is located 0 bytes to the right of 8-byte region [0x60040000dfb0,0x60040000dfb8)
allocated by thread T0 here:
    #0 0x7fcb1ccfa55f (/usr/lib/x86_64-linux-gnu/libasan.so.0.0.0+0x1555f)
    #1 0x7fcb1c9763c2 (/usr/lib/libsoletta.so.0.0.1+0x10e3c2)
    #2 0x7fcb1c8f6008 (/usr/lib/libsoletta.so.0.0.1+0x8e008)
    #3 0x7fcb1c8f693f (/usr/lib/libsoletta.so.0.0.1+0x8e93f)
    #4 0x7fcb1c8f6a1d (/usr/lib/libsoletta.so.0.0.1+0x8ea1d)
    #5 0x7fcb1c95f2ee (/usr/lib/libsoletta.so.0.0.1+0xf72ee)
    #6 0x7fcb1c95fed7 (/usr/lib/libsoletta.so.0.0.1+0xf7ed7)
    #7 0x7fcb1c96040c (/usr/lib/libsoletta.so.0.0.1+0xf840c)
    #8 0x7fcb1c94d7c8 (/usr/lib/libsoletta.so.0.0.1+0xe57c8)
    #9 0x7fcb1c960d97 (/usr/lib/libsoletta.so.0.0.1+0xf8d97)
    #10 0x7fcb1c94c8ea (/usr/lib/libsoletta.so.0.0.1+0xe48ea)
    #11 0x40c7e6 (/usr/bin/sol-fbp-generator+0x40c7e6)
    #12 0x7fcb1c2a6ec4 (/lib/x86_64-linux-gnu/libc-2.19.so+0x21ec4)
Shadow bytes around the buggy address:
  0x0c00ffff9ba0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9bb0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9bc0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9bd0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9be0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
=>0x0c00ffff9bf0: fa fa fa fa fa fa 00[fa]fa fa 00 00 fa fa 00 fa
  0x0c00ffff9c00: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9c10: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9c20: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9c30: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c00ffff9c40: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:     fa
  Heap righ redzone:     fb
  Freed Heap region:     fd
  Stack left redzone:    f1
  Stack mid redzone:     f2
  Stack right redzone:   f3
  Stack partial redzone: f4
  Stack after return:    f5
  Stack use after scope: f8
  Global redzone:        f9
  Global init order:     f6
  Poisoned by user:      f7
  ASan internal:         fe
==466== ABORTING

edersondisouza, 2015-09-01:
On pr #668 there's a patch that should fix this issue (05cd23815 sol-util-file: Fix sol_buffer_at_end usage).
As stated on commit, `sol_buffer_at_end()` has an off by one issue, thus the invalid read of '0 bytes to the right of region...'. As `sol_buffer_at_end()` has more uses that expect this off by one, I've fixed locally on `sol_util_file_load_string()`.

edersondisouza, 2015-09-02:
@lgywata PR #668 has been integrated. Please check if this issue has been solved.

lgywata, 2015-09-14:
@edersondisouza: yes, it's solved. Thanks.

closed 2015-09-14 by lgywata