solettaproject/soletta #747: LCD (String) + Buttons high-level node types

barbieri, 2015-09-17:
It is common to have IoT products with LCD display with buttons (arrows) and need to ask for user input. We should offer some high-level node types to deal with that, for instance a "selector" as required by #732 (see that issue for description on a suggested node type).

Generally speaking these node types should format a multi-line string to be used by other node types such as the `grove/lcd-string`. This is done by taking the number of rows and columns in that display and doing the formatting. 3 input ports should be used to control the component (like: previous/up, next/down and ok/select).

At least the following components should exist:
 * **Boolean:** enter true or false (on/off, labels to come as options).
 * **Integer:** enter an integer value. Input ports increase/decrease the value within a range.
 * **Integer-Large:** similar to integer above, but meant for large numbers like years. This should use 2 more ports to change the position, while increase/decrease ports will affect only that digit at given position.
 * **String:** enter a string. Similar to *integer-large* above, two ports to change the position, two to cycle chars. An option should specify the characters to allow, defaults to ASCII printable.
 * **String-formatted:** similar to string, this takes a format and allows editing "fields" from it. For instance a format could be "{3d:0,255,1}.{3d:0,255,1}.{3d:0,255,1}.{3d:0,255,1}" to enter an IP address or "{2x}:{2x}:{2x}:{2x}:{2x}:{2x}" to edit MAC Address. The node would only edit the fields contained inside it, skipping the formatting such as "." and ":" in above examples. This could be the base for some helpers such as timestamp editor.
 * **Selector:** selects from a list of items, similar to HTML's `<select>`

Nodes should be similar in ports and options. From #732 we could take the proposed interface for **selector**:
   * Options:
       * `columns` (int): the number of columns in the display. Will be used to format the string, including auto-scroll.
       * `rows` (int): the number of rows in the display. If more than 1, will wrap lines.
       * `title` (string): if more than one row, this will be the contents of the first row. Otherwise will be the prefix of the only row in the string.
       * `auto_scroll` (boolean): if true and the contents doesn't fit in the columns, then each row will scroll.
       * `circular` (boolean): if true, this is a circular menu, after the last option it goes back to the first.
    * Input Ports:
       * `ITEMS` (custom packet): the options to be selected.
       * `NEXT` (any): change to the next item.
       * `PREVIOUS` (any): change to the previous item.
       * `SELECT` (any): select the current item.
       * `ENABLED` (bool): make the interaction possible, otherwise don't produce strings or process next/previous/select inputs. This is used because the node is not created and destroyed on demand, rather it will be enabled and disabled.
       * `SELECTED` (string): the previously selected item, if any.
    * Output Ports:
       * `SELECTED` (string): the selected identifier. Sent when user chooses via HTTP.
       * `STRING` (string): string with the contents, this can be fed to LCD displays such as `grove/lcd-string.

Of particular interest is the control of the output string, taking a `title` (or some format, that users could include the title), number of `columns` and `rows`, as well as `auto_scroll`. Auto scroll and title or format should be linked and try to fit well the user needs: no scroll (crop), scroll the whole row or only the part where user enter/select contents.

I rather go the `format` route and allow stuff such as:  `format="Size: {value} px"` for a single lined displays or `format="Size:\n{value} px"` for two lines. If we use `title`, then for integers we'd need both `suffix` and `prefix`, so one could better present information to user. With format we could use some of the string formatting syntax and allows alignment to be specified as well.

**NOTE:** this issue is an umbrella, we don't need to implement all of them for v1. Implement the simple nodes first and then create new issues for remaining nodes.

glima, 2015-09-28:
Nice, I'm now scratching a first go. What's the point/function of the *input* SELECTED port, BTW?

barbieri, 2015-09-28:
say you persisted the last user selected item, then user wants to change it, how to show what was selected? You send that item (or id) on the SELECTED port.

glima, 2015-11-06:
For the form/string node, I wonder how to define the length of the string in formation. Do we allow growing it only, whenever the cursor goes past it rightwards? Or do we allow to remove the last char (if so, how to indicate it differentiating from a space character)?

barbieri, 2015-11-06:
add a "BACKSPACE" (or DELETE_BACKWARDS") that deletes the previous char. Or a "DELETE" that deletes the current char (making no empty space).

glima, 2015-11-06:
Okay, going for it.

glima, 2015-11-12:
For the string-formatted node I'll ditch any form of input SELECTED port, OK?

barbieri, 2015-11-13:
no need "SELECTED", but need the existing contents? Or am I missing something?

glima, 2015-11-13:
I mean it'll be harder to get initial contents from a port. We can have the port and do the best trying to match the syntax of what's expected and what comes in, but I'll do that after I have the basics working.

barbieri, 2015-11-13:
This is more of use to load saved values. You can simply check if it is in the format, if not reject the packet.

glima, 2015-11-17:
Got to it now, and it's hard: think a format with neighbor numeric entries, like ``{d:0,255,1}{d:0,255,1}blah``. It's impossible to know before hand how many digits the numeric fields will have (1 to 3), so parsing is cumbersome.

glima, 2015-11-17:
Ok, I think I'll leave it with the check "if any neighbor entries are both numeric in the format, don't even bother to parse, otherwise proceed". The numerical formats accept any of Python's now (doubles included). I'm wondering if a (maybe optional) ``<val>`` spec deserves to join the whole ``{<type>:<min><max><step>}`` construction.

glima, 2015-12-01:
Done, awaiting review.

closed 2015-12-01 by glima