solettaproject/soletta #748: HTTP Client generic node types

barbieri, 2015-09-17:
We should provide some generic node types to deal with HTTP/REST APIs that we lack native/optimized components. Ideally we should be as simple as `wget | sed` to solve simple problems, linking with `string/regexp-search`, `string/regexp-replace` and the node types from #746.

Suggested node types:
 * `http/get-string`: does a `GET` and outputs a string packet.
    * Options:
      * `url` (string): the URL to execute `GET`.
      * `strict` (boolean, default=false): validate `Content-type` is same as requested.
      * `content_type` (string, default="text/plain"): provided as `Accept: ` and then validated with `Content-type:` HTTP headers.
    * Input Ports:
      * `URL` (string, optional): allows changing the `url` given as option.
      * `TRIGGER` (any): execute the HTTP method.
    * Output Ports:
      * `OUT` (string): the string returned as payload of the HTTP request.
 * `http/get-json`: does a `GET` and outputs a json object packet.
    * Options:
      * `url` (string): the URL to execute `GET`.
      * `strict` (boolean, default=false): validate `Content-type` is same as requested.
      * `content_type` (string, default="application/json"): provided as `Accept: ` and then validated with `Content-type:` HTTP headers.
    * Input Ports:
      * `URL` (string, optional): allows changing the `url` given as option.
      * `TRIGGER` (any): execute the HTTP method.
    * Output Ports:
      * `OUT` (json object): the json object returned as payload of the HTTP request.
 * `http/get-blob`: does a `GET` and outputs a blob packet.
    * Options:
      * `url` (string): the URL to execute `GET`.
      * `strict` (boolean, default=false): validate `Content-type` is same as requested.
      * `content_type` (string, default=null): if non-null, provided as `Accept: ` and then validated with `Content-type:` HTTP headers.
    * Input Ports:
      * `URL` (string, optional): allows changing the `url` given as option.
      * `TRIGGER` (any): execute the HTTP method.
    * Output Ports:
      * `OUT` (blob): the blob returned as payload of the HTTP request.
 * `http/request`: creates a HTTP request to the server (complex version that covers all cases):
    * Options:
      * `url` (string): the URL to execute method.
      * `method` (string): one of `GET`, `POST`, `HEAD`, `PUT`...
      * `timeout` (int, default=0): the number of milliseconds to wait for this request to be completed.
      * `allow_redir` (boolean, default=true): allows redirections to be followed.
    * Input Ports:
      * `URL` (string, optional): allows changing the `url` given as option.
      * `METHOD` (string, optional): allows changing the `method` given as option.
      * `TIMEOUT` (integer): allows changing the `timeout` given as option.
      * `PARAM_QUERY` (key-value): appends query parameter.
      * `PARAM_COOKIE` (key-value): appends cookie parameter.
      * `PARAM_POST` (key-value): appends post parameter.
      * `PARAM_HEADER` (key-value): appends header.
      * `USER` (string): the username to authenticate.
      * `PASSWORD` (string): the password to authenticate.
      * `CONTENT_TYPE` (string): the content-type of the payload (goes in header).
      * `CONTENT` (blob): the payload to send with the method.  
      * `TRIGGER` (any): execute the HTTP method.
    * Output Ports:
      * `OUT` (http-request): result of the HTTP request.

Where the `HTTP_REQUEST` packet type is composed of:
 * `RESPONSE_CODE` (int): the response code such as 200 (success) or 404 (not found).
 * `CONTENT_TYPE` (string): the content type of the payload.
 * `URL` (string): the request URL.
 * `CONTENT` (blob): the request content.
 * `PARAMS_QUERY` (array of key-value): 
 * `PARAMS_COOKIE` (array of key-value): 
 * `PARAMS_POST` (array of key-value): 
 * `PARAMS_HEADER` (array of key-value): 

Then specific node-types to extract information from `HTTP_REQUEST`, such as `request-get-url`, `request-get-param-query`, etc.

otaviobp, 2015-10-01:
Note that I sent a patch adding JSON Object packet types in #818

cabelitos, 2015-10-23:
After talking to @bdilly @ceolin and @otaviobp We concluded that for now the get-json node is not necessary. As one can easily obtain a json object using the http-client/get-blob -> converter/blob-to-json-object nodes.

barbieri, 2015-10-23:
IMO wouldn't hurt and save people in the common case, that is JSON web APIs

bdilly, 2015-10-26:
It isn't not hard to create / maintain a new node type to do so, since it's a matter of adding just a new process() function, to validate and send json packets.
It won't be a single http/get-json with a single OUT port, since it may be both objects or arrays.

Initially I was more inclined to just have a get-blob and converters to string, json-object and json-array.
No big deal.

@cabelitos is going to add these node types and also some more tests / samples tomorrow so we can close this task.

barbieri, 2015-10-27:
It should be a single one, JSON_OBJECT as the root should always be an object for these API. if users are going non-standard they can use blob+conversions (as if the guy uses a string/number/true/false/null) he needs to do other stuff anyways. Or we auto detect and provide multiple out ports, one for each (like object-get)

cabelitos, 2015-10-27:
@barbieri I created a node that has two outputs. json-object and json-array. 

According to https://www.ietf.org/rfc/rfc4627.txt a json-text (the response from an http request as json) can be an object or an array, both are valid.

See #969

cabelitos, 2015-10-28:
All nodes are now merged. Closing this task.

closed 2015-10-28 by cabelitos