solettaproject/soletta #914: Flow/HTTP: Add support for Tago

lpereira, 2015-10-19:
Similar to Thingspeak, Tago allows people to send and/or receive data to the cloud so that they can be visualized and acted upon. Implement two node types (one to send data, other to retrieve), implementing its straightforward API: https://www.tago.io/en/developers/api/introduction

(The Freegeoip node type and location packet type have been implemented so that implementing this would be easier.)

Bonus points if there's an option to set the API endpoint, the same way that the Thingspeak nodes do. This allows people to implement their own Tago instance if needed be.

bdilly, 2015-11-04:
@lpereira regarding this bonus point, are you sure it's an open source project, in the sense you can run your own instance? I only found repositories of clients for Tago.

Indeed API seems to be really straightforward

lpereira, 2015-11-04:
[**@bdilly**](https://github.com/bdilly) commented on [4 Nov 2015, 20:25 GMT-2](https://github.com/solettaproject/soletta/issues/914#issuecomment-153888868 "2015-11-04T22:25:49Z - Replied by Github Comment Enhancer"):
> @lpereira regarding this bonus point, are you sure it's an open source project, in the sense you can run your own instance? I only found repositories of clients for Tago.
> 
> Indeed API seems to be really straightforward

It's not open source, but given that the API is straightforward, I don't see why it couldn't be easily implemented (without all its bells and whistles, of course) as an open source project. In fact, I suggested that our HTTP server nodes exposed the same API but in the end it was decided we should go to another direction.