solettaproject/soletta #962: string-pcre-regexp-search.fbp failed to run on Galileo gen2 board

dborquez, 2015-10-26:
version: v1-beta8 (sha1: 114cf120a7f4720c91e8ce1f5b3bdb7020694888)

- Output returned:
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'result_01' failed

- Complete output returned: http://pastebin.intel.com/view/e273fb74

anselmolsm, 2015-10-27:
Please, use a public pastebin service in public reports (e.g. https://gist.github.com, http://pastebin.com). This also applies to #963, #964 and #965. Tks.

barbieri, 2015-10-27:
who compiled that libpcre? It's missing features, like the log says. It's not a soletta bug.

dborquez, 2015-10-27:
The libpcre installed not satisfy the version >=8.12
- libpcre-dev - 8.35-r0.0
- libpcre1 - 8.35-r0.0
- libpcrecpp0 - 8.35-r0.0
- libpcreposix0 - 8.35-r0.0

bdilly, 2016-02-11:
It was an issue with libpcre missing proper utf8 support.

It requires --enable-utf8 and --enable-unicode-properties

Fixed by https://github.com/solettaproject/meta-soletta/pull/33

closed 2016-02-11 by bdilly