solettaproject/soletta #963: persistence-fs-1 failed to run on Galileo Gen2

dborquez, 2015-10-26:
version v1-beta8 (sha1:  114cf120a7f4720c91e8ce1f5b3bdb7020694888 )

- output returned:
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node '#anon:143:94' failed

- complete output returned:
http://pastebin.intel.com/view/0baba05b

barbieri, 2015-10-27:
the log is related to another bug (javascript), please provide the paste for persistence failure.

Also, if you want you can open a Markdown block in your issue description. Use triple braces: `{{{` to start and `}}}` to end (each on its own lines)

edersondisouza, 2015-11-12:
Soletta Galileo image [v1_beta11](https://github.com/solettaproject/meta-soletta/releases/tag/v6) and later should have this issue fixed. It is related to compilation flags.

closed 2015-11-12 by edersondisouza