solettaproject/soletta #965: [P2] javascript.fbp test failed on Galileo Gen2

dborquez, 2015-10-26:
version: v1-beta8 (sha1 114cf120a7f4720c91e8ce1f5b3bdb7020694888 )

- Output returned:
ERR:sol-flow ./src/lib/flow/sol-flow-parser.c:706 parse_declarations() Couldn't handle declaration 'MyScript' of metatype 'js'

- Complete output: 
http://pastebin.intel.com/view/34b7ef1c

barbieri, 2015-10-27:
who compiled soletta? did you use git submodules to download `duktape`? It is missing and thus was not compiled (`js.so`)

dborquez, 2015-10-27:
I was not aware of that, because I based the build in the documentation available for the galileo:
https://github.com/solettaproject/soletta/wiki/Galileo-and-Galileo-Gen2-Instructions

After duktape and other submodules were installed the issue reported here has disappeared.

Thank you.

closed 2015-10-27 by dborquez

barbieri, 2015-10-27:
@mbelluzzo or @anselmolsm, could you update the build instructions to reflect submodules usage?

anselmolsm, 2015-10-27:
Added to https://github.com/solettaproject/soletta/wiki/Galileo-and-Galileo-Gen2-Instructions and  https://github.com/solettaproject/soletta/wiki/Build-requirements.