solettaproject/soletta #1006: calamari-7seg-segments.fbp, calamari-led.fbp and calamari-buttons-rgb-led.fbp failed to run on minnowboard

dborquez, 2015-10-30:
**Soletta version:** V5
**sha1:** ed93a6f8d1d572f1e67da78651be268758f93631

**Output details:**
https://gist.github.com/dborquez/ce1a3fa5d4f2b77c346f

Thank you.

bdilly, 2015-12-07:
@brunobottazzini could you please close it if was fixed already? I remember you mentioned calamari demos are working with V6.

brunobottazzini, 2016-02-23:
@bdilly we had tested it and we got problems only in the samples:

 - calamari-7seg-segments
 - calamari-button-accumulator-persistence
 - calamari-buttons-rgb-led

These problems should be fixed with the PR: https://github.com/solettaproject/soletta/pull/1462.

For the next release everything should be normal.

bdilly, 2016-02-23:
cool.
so I'm closing this issue

closed 2016-02-23 by bdilly