solettaproject/soletta #1062: sol-fbp-generator could use duktape's bytecode dump/load

barbieri, 2015-11-15:
meta-type "js" should pre-parse the given JS and use `duk_dump_function()` (see https://github.com/svaarala/duktape/blob/master/doc/bytecode.rst) and then include the bytecode as a `static const uint8_t *`. The generated code should then output an `open()` that uses `duk_load_function()` to read from the same variable.

Ideally, this would allow to use JS on systems without filesystem, such as Contiki and RIoT, given that they have enough memory to run Duktape (see https://github.com/svaarala/duktape/blob/master/doc/low-memory.rst)

**NOTE:** The bytecode is not portable for different versions of Duktape. We should generate code that copes with that, eventually by using `DUK_VERSION` (https://github.com/svaarala/duktape/blob/master/src/duk_api_public.h.in#L83 on details of the encoding) from the generator and the compile time. That is:
 * sol-fbp-genator outputs the `DUK_VERSION` it was compiled to, stating that the bytecodes are available in that version;
 * the generated.c will check at runtime (if dynamic modules or shared library are enabled) if the versions match, following with the bytecode. Otherwise fallsback to load from file. Do this with `#ifdef` in a way that statically-compiled binaries or systems without FS will not have the fallback compiled in.

This is low priority, just a reminder to look at it later. Assigned to @cabelitos since he did work with `sol-fbp-generator` and meta types, but anyone could take it.

cabelitos, 2015-11-26:
@barbieri 
Hello,

I don't remember with I talked to you in person about this or no. Anyway I will ask again.
I made the first implementation for the js meta type code dump. The generated code does not rely on open() syscalls, because the whole JS code is dumped in the generated C file, then duk_eval_string(TheWholeJSCodeHere) is used.

I would like to know if you still prefer the byte code solution.

barbieri, 2015-11-26:
The bytecode would be good to keep as a first-try.

It would be nice if we could `#ifdef` that block that falls back to the actual string (what you have now), make it available only if Soletta was built with dynamic library -- thus for small OS or the micro-linux case (libsoletta.a instead of libsoletta.so) we can throw away the script as text, keeping only the bytecode.

I'm not sure we have that information in a public header such as sol-common-buildopts.h, if not we should add it there and check in the generated ".c"

cabelitos, 2015-12-09:
Please look at: https://github.com/solettaproject/soletta/pull/1167

cabelitos, 2015-12-16:
@barbieri Should we close this issue?

barbieri, 2015-12-16:
ok, investigated and is not worth at the moment. See #1167 for future references (please do not delete that PR or branch)

closed 2015-12-16 by barbieri