solettaproject/soletta #1084: flow/http-{client,server} support for composed packet types

barbieri, 2015-11-19:
http-client and server should both support composed packet types, allowing more complex data structures to be shared.

For simple internal types it is easy to support in both JSON and POST or get query parameters:
 * json: ["a", 1, false]
 * query: value=a&value=1&value=false 

For nested, JSON is much easier, but I'm not sure we support those elsewhere
 * json: ["a", [1, false]]
 * query: would need to define some way to recurse, like: value=a&value=value%3d1%26value%3dfalse

How to expose those as nodes is also something to think about. We could offer meta-types to create them, that guarantees proper types are defined, validated and exchanged, or we may offer to do nodes with ports "any", this would check the packet type and do the best effort to encode to network, with the parse from network outputing something like JSON_ARRAY...

I like the first case more, to expose as a meta-type.

cabelitos, 2015-11-20:
@barbieri Hello, Why the input must be in json or HTTP query form? The user could easily use the implemented meta types to create composed packets. All we have to do is implement http client/server nodes that can handle them.

barbieri, 2015-11-20:
not sure I get it, clarify?

I was thinking alongside what we provide for basic types, such as the boolean, irange, string... if the user wants to send/receive a batch, like user + password, then he could use a composed and then this node, not requiring him to do any further conversion (ie: no composed -> split -> json -> blob -> http)

ceolin, 2016-04-14:
both client and server were commited.

closed 2016-04-14 by ceolin