solettaproject/soletta #1125: Public headers need to be buildable using g++

gabrielschulhof, 2015-11-27:
When using g++ to compile a file which includes soletta public headers, it dies on code like

```
In file included from /home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-flow-packet.h:40:0,
                 from /home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-flow.h:40,
                 from /home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-flow-static.h:35,
                 from ../generated/main.h:8,
                 from ../generated/main.cc:7:
/home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-str-slice.h: In function ‘sol_str_slice sol_str_slice_from_blob(const sol_blob*)’:
/home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-str-slice.h:68:88: warning: invalid conversion from ‘void*’ to ‘const char*’ [-fpermissive]
 #define SOL_STR_SLICE_STR(_s, _len) (struct sol_str_slice){.len = (_len), .data = (_s) }
                                                                                        ^
/home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-str-slice.h:146:12: note: in expansion of macro ‘SOL_STR_SLICE_STR’
     return SOL_STR_SLICE_STR(blob->mem, blob->size);
            ^
```

The above is only a warning and not an error because I passed ```-fpermissive``` to g++, but I'd rather pointers were cast than have to add this flag.

More importantly, there's
```
In file included from /home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-i2c.h:40:0,
                 from ../generated/main.h:57,
                 from ../generated/main.cc:7:
/home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-buffer.h: At global scope:
/home/nix/iot/soletta-js/depbuild/soletta/build/soletta_sysroot/usr/include/soletta/sol-buffer.h:279:125: error: expected primary-expression before ‘static’
 int sol_buffer_insert_as_base64(struct sol_buffer *buf, size_t pos, const struct sol_str_slice slice, const char base64_map[static 65]);
                                                                                                                             ^
```

for which I cannot think of a flag to pass to g++ to get it to accept the code.

cmarcelo, 2015-11-27:
It seems this is a C feature that doesn't have equivalent in C++. If that's the case we could use a macro so that in C++ the replacement would be empty.

@lpereira thoughts on whether C++ has something equivalent?

barbieri, 2015-11-27:
I think @cmarcelo proposal for macro in the "static N" is the alternative, something like SOL_STATIC_ARRAY_SIZE(n)   (I'm okay with other names, but remember to keep the SOL prefix.

As for the first error in buffer, it would be better to have the macro to use GCC type checking expression before doing a cast -- or have the cast explicitly in the call site.

Last but not least, would be nice to have a test that includes all our installed headers and compile with g++ to ensure we're not breaking in future commits.

cmarcelo, 2015-11-27:
For the record, @lpereira pointed that lwan (his project) uses a macro as well https://github.com/lpereira/lwan/blob/master/common/lwan.h#L101

cmarcelo, 2015-11-27:
@gabrielschulhof a patch was applied for this issue, could you test again with master?

gabrielschulhof, 2015-11-29:
With master as of f3aa5fb82145e6dfc08a9ab58c5aa0c93a1fbe55 I can't get soletta itself to build:

```
./src/lib/comms/sol-socket-dtls-impl-tinydtls.c: In function ‘sol_socket_dtls_set_handshake_cipher’:
./src/lib/comms/sol-socket-dtls-impl-tinydtls.c:836:64: error: ‘TLS_ECDH_anon_WITH_AES_128_CBC_SHA_256’ undeclared (first use in this function)
         [SOL_SOCKET_DTLS_CIPHER_ECDH_ANON_AES128_CBC_SHA256] = TLS_ECDH_anon_WITH_AES_128_CBC_SHA_256,
                                                                ^
```

... so, I'll wait for a commit on soletta's master which will allow me to build it, before I can check whether the headers can be included from C++.

bdilly, 2015-12-17:
@ceolin could you please take a look on this issue and see if we can close it?
It seems the case for me

ceolin, 2015-12-18:
yep,  i'll see it.

ceolin, 2015-12-21:
i've checked that many headers are not buildable with g++.  Those headers contains inline code that need to be changed, things like implicit cast:

error: invalid conversion from ‘void ’ to ‘const char*’ [-fpermissive]

I'm going to review our headers to fix this issue.

ceolin, 2015-12-22:
I've fixed our public headers to build using c++ compiler. It's required c++0x support, so it's necessary `-std=gnu++0x` or `-std=c++0x` option.
Please check https://github.com/solettaproject/soletta/pull/1219

ceolin, 2015-12-28:
as the pr was merged and nobody complaint, i'm closing this issue. Please feel free to open if it happens again.

ceolin, 2015-12-28:
pullrequest was merged, the problem seems to be solved

closed 2015-12-28 by ceolin