solettaproject/soletta #1134: Custom nodes will fail to run if it depends on a Soletta node modules.

cabelitos, 2015-11-30:
If one creates a custom node that depends on flow modules that are not built as built-in, the flow will fail to run.

We have the follolwing scenario:
SML creates a node called message-constructor, which is responsible for grouping flower-power data relavant to SML. The message-constructor depends on the flower-power module which is built as module by default.
However when running the flow the flower-power module is not loaded at all, thus the sol-fbp-runner will complain and the execution will be aborted.

We should create a way to specify that a node X depends on node Y in order to load its .so file.

Note: A quick fix for the problem is build the module as built-in.

Check: https://github.com/solettaproject/soletta-machine-learning/issues/62

https://github.com/solettaproject/soletta-machine-learning/blob/master/soletta_module/sml_garden/sml_garden.c

https://github.com/solettaproject/soletta-machine-learning/blob/master/soletta_module/samples/sml_garden/sml_garden.fbp

barbieri, 2015-12-01:
If you need the node types, then it should get via flow resolver, as usual.

If you need other symbols, use @ibriano's lib loader (https://github.com/solettaproject/soletta/blob/master/src/shared/sol-lib-loader.h), it should be used through subsystem specific wrappers such as https://github.com/solettaproject/soletta/blob/master/src/lib/flow/sol-flow-modules.c (see macro in sol-flow.h)

We may need to introduce one to get the packet type and add that to sol-flow-packet.h, using to get symbols from built in or external

cabelitos, 2015-12-01:
@barbieri I think that this load should be done automatically by Soletta. As a user, I don't think I need to worry about that, I just want to use my custom node without needing to worry that I need to load the libraries X,Y,Z.
What do you think?

barbieri, 2015-12-01:
I agree, that's why I'm asking what it is using. If it is node-by-typename, then it should happen, if not it's a bug.

If it's creating a packet-type, then the pointer to that symbol should be acquired as I said, by the tool/generators

cabelitos, 2015-12-01:
Sorry, I didn't understand that it was a question. I'm trying to get the message contents
with sol_flower_power_get_packet() (which is exported by the flower power module)

barbieri, 2015-12-01:
explained in person, but to repeat it here:

If all we need is a `struct sol_flow_packet_type*`, then we can export a function/macro similar to `sol_flow_get_node_type()` and `sol_flow_internal_get_node_type()`, but target at packets. This is mostly about types/casts, no logic in it, see https://github.com/solettaproject/soletta/blob/master/src/lib/flow/sol-flow-modules.c

That mean we couldn't use symbols such as `sol_flower_XXX`, but we could get the packet and use base functions such as `sol_flow_send_packet()`, `sol_flow_packet_type_new()` and so on.

If we want to expose symbols in general, we can add yet another set of macros/function, but they won't guarantee types, people will have to deal with `void *`

ceolin, 2015-12-15:
@barbieri  anyway who is "exporting" a custom packet will need to give the description of the packet's data right ? In this case, `flower_power` should give a header with the packet data struct

barbieri, 2015-12-15:
header files are solved at compile time, not at runtime. They won't try to resolve any symbol.

We should use `data_size` to verify if we have the same amount of bytes before we continue (it could be a one time check at the entry point).

If we think it's needed we may introduce an extra `sub_api` in packet_type, so we can report if that packet changed it ABI, allowing runtime checks.

ceolin, 2015-12-15:
I was not saying that the header would solve the symbol's problem. I wanted to say that even resolving the packet's type symbol, it'll be necessary who is exporting the packet also gives the packet's data struct, otherwise it'll be an amount of bytes without any meaning for the user. This is what is done by flower-power.

Anyway, i've already solved this problem in the simplest way (creating the `sol_flow_get_node_type()` equivalent for packets). My concern is, the user will need to know the module's name.

barbieri, 2015-12-16:
ok

ceolin, 2015-12-16:
the fixes were pushed in both soletta and soletta-machine-learning.

closed 2015-12-16 by ceolin