solettaproject/soletta #1148: Add API to change OIC server device information

otaviobp, 2015-12-02:
Some device information are hardcoded in soletta. Create API to let users change that.

barbieri, 2015-12-02:
like what? I recall some were specified in terms of define when soletta was compiled.

If we could avoid set/get apis to do these things, then great. Like if we can use device-id from machine-id, product name/version from some sol-platform (maybe they don't exist yet, but should) that get from system-specific places (like /etc/os-release on linux)

lpereira, 2015-12-02:
Some of these parameters are already obtained from the system using the APIs provided by Soletta (machine ID, operating system, version, etc); these, in fact, should not be available to be changed by the user. But there are some others that should be able to be modified if the user is not building Soletta specifically for their application, or if a binding to another language is to be produced, such as the device name, its version, support URI, etc.

otaviobp, 2015-12-04:
I updated the description adding which fields needed to be available using the API

otaviobp, 2016-05-12:
I added in kbuild config a way of changing this information. Do we need to have an API for changing that too? Is there any use case for that?

barbieri, 2016-05-12:
at least some fields would be reasonable to change in runtime, for instance in Ostro we'd compile the libsoletta at one stage and users creating products would only compile their app, then they must provide a runtime value.

What if we convert these to `struct sol_oic_server_info { const char *x, *y, *z...}` and make this an optional member (pointer) of the constructor/config? Then if provided, this is used, otherwise the global is used. The definition of the global is very simple, just wrap the global variables in a struct.

We can document this struct must stay alive during oic server lifetime, then there is no need to copy them.