solettaproject/soletta #1149: Cover 100% public API (headers / samples) on documentation

glima, 2015-12-02:
Don't forget the samples -- we need much more text explaining what's going on each of them for newcomers.

mbelluzzo, 2015-12-07:
I think its wise to split c-api docs (headers) from samples docs, specially because they are created in a different way and many of our samples may need re-work.

glima, 2015-12-07:
That's OK. Would you edit this ticket and create the other for the samples, please?

Gustavo Lima Chaves
Software Engineer @ Intel

mbelluzzo, 2015-12-07:
Sure

mbelluzzo, 2015-12-07:
Soletta Lib (considering my pr and bdilly's http):
**crypto** = 100%
**datatypes** = 100%
**io** = 100%

**comms** = 64%
**common** = 41%
**parsers** = 17% - WIP is a single file
**flow** = 16%

bdilly, 2015-12-28:
@mbelluzzo concluded this task.

I've added checks on semaphoreci to avoid merges of new API without documentation.

This way we'll be sure to always have the whole API covered.

closed 2015-12-28 by bdilly