solettaproject/soletta #1158: HTTP-server fbp tests still failing with timeout

glima, 2015-12-04:
On Koji, Fedora's RPM build system, when building packages for Fedora rawhide (24 WIP), which has 0.9.46-1.fc24 as version for libmicrohttpd, we get

```
+ make 'CFLAGS= -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m32 -march=i686 -mtune=atom -fasynchronous-unwind-tables' 'LDFLAGS= -Wl,-z,relro -specs=/usr/lib/rpm/redhat/redhat-hardened-ld' -j4 check-fbp
FAILED running http-boolean.fbp: wrong exit code, got 1 but expected 0
Output:
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:921 client_request_internal() Could not set POST fields or data from params
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:996 sol_http_client_request() pending== NULL
WRN: ./src/modules/flow/http-client/http-client.c:440 common_post_process() connection== NULL
WRN:sol-flow-test ./src/modules/flow/test/result.c:83 on_timeout() Timeout expired! Failing test...
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'test_server_output' failed
FAILED running http-drange.fbp: wrong exit code, got 1 but expected 0
Output:
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:921 client_request_internal() Could not set POST fields or data from params
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:996 sol_http_client_request() pending== NULL
WRN: ./src/modules/flow/http-client/http-client.c:440 common_post_process() connection== NULL
WRN:sol-flow-test ./src/modules/flow/test/result.c:83 on_timeout() Timeout expired! Failing test...
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'test_server_output' failed
FAILED running http-irange.fbp: wrong exit code, got 1 but expected 0
Output:
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:921 client_request_internal() Could not set POST fields or data from params
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:996 sol_http_client_request() pending== NULL
WRN: ./src/modules/flow/http-client/http-client.c:440 common_post_process() connection== NULL
WRN:sol-flow-test ./src/modules/flow/test/result.c:83 on_timeout() Timeout expired! Failing test...
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'test_server_output' failed
FAILED running http-string.fbp: wrong exit code, got 1 but expected 0
Output:
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:921 client_request_internal() Could not set POST fields or data from params
WRN: ./src/lib/comms/sol-http-client-impl-curl.c:996 sol_http_client_request() pending== NULL
WRN: ./src/modules/flow/http-client/http-client.c:440 common_post_process() connection== NULL
WRN:sol-flow-test ./src/modules/flow/test/result.c:83 on_timeout() Timeout expired! Failing test...
ERR:sol-flow-test ./src/modules/flow/test/result.c:74 fail() test/result node 'test_server_output' failed
4 tests failed.
186 tests passed.
tools/build/Makefile.targets:8: recipe for target 'check-fbp' failed
make: *** [check-fbp] Error 1
RPM build errors:
error: Bad exit status from /var/tmp/rpm-tmp.1VJDpK (%check)
    Bad exit status from /var/tmp/rpm-tmp.1VJDpK (%check)
```

as build output. It seems we still have bugs either on Soletta or the HTTP server side.

ceolin, 2015-12-09:
it not seems to be the timeout problem, it's failing to set the post fields. I'll investigate it.
Please, in further issues/problems add the curl/microhttpd version.

glima, 2015-12-10:
I just missed curl's version, I'll see if I can get it yet.

glima, 2015-12-10:
``DEBUG util.py:393:   curl                        i686   7.46.0-2.fc24                   build 292 k``

ceolin, 2015-12-11:
thanks, it's happening all the time ? 
This is definitly not the server's timeout problem, is a problem when setting the post fields. I could not reproduce in my machine, i'll try reproduce it in an environment more similar.

glima, 2015-12-14:
Mm, maybe if you test with the same curl's version, then?

ceolin, 2015-12-16:
i don't think so, something is goind wrong when setting the postfields and/or postdata. Probably this is something related with 32 bits arch since none of us could reproduce it. Iscaro is going to build it in 32bits to reproduce the problem

ceolin, 2015-12-16:
@cabelitos please close this issue as soon as you push the fix

cabelitos, 2015-12-16:
@ceolin No problemo.

cabelitos, 2015-12-16:
Merged a fix commit ef29993

closed 2015-12-16 by cabelitos