solettaproject/soletta #1198: Build bots should compile also Soletta in 32-bits

cabelitos, 2015-12-16:
I'm not quite sure if someone already brought this up, anyway I will describe it here.

Right know Soletta's build bots are generting 64 bits build, since they're running on 64 bit machines.
However as we support 32-bits archs, we should also build Soletta using 32-bits, and the most important part, run all the tests.

This may sound silly, but only today we fixed two bugs that was only happing on 32-bit machines.
(check ef29993d1b672519104a6caf5f7e9b0b63b9b59b and https://github.com/solettaproject/soletta/pull/1196)

Suggestions @barbieri @bdilly @ceolin @glima @edersondisouza @otaviobp ?

barbieri, 2015-12-17:
+2, running linux tests should be easy, usually these systems ship with multilib, but we can also use qemu and test small OSes and linux we can use to test linux-micro platform.

just need someone to implement a wrapper to execute these on qemu, we can do that inside our "make check" rules. The cross compile can happen with `TOOLCHAIN_PREFIX=`

cabelitos, 2015-12-17:
@barbieri +2 for small Oses as well!

ceolin, 2015-12-17:
would be nice build on arm as well

glima, 2015-12-17:
Qemu sounds interesting, indeed. The 1st 32-bit-only but was spotted by a Fedora-ARM bot, BTW.

bdilly, 2015-12-17:
uhm... doing this on semaphoreci seems to be a bit problematic, since it runs all the setup from a initial vm image for every pull request, so I suppose it would increase a lot the pull request validation time.

on tipbot, in the other hand, we could let some images for these different platforms configured already, and the overhead would be only building soletta and running tests again on each of these images.