solettaproject/soletta #1247: Add support for NuttX

acassis, 2015-12-30:
NuttX is a RTOS used in many projects (PX4 drones control board, ThinkSee, Google Project Ara, etc). More info: https://en.wikipedia.org/wiki/NuttX

barbieri, 2016-01-05:
Hi Alan,

I'm glad to see you around, do you have interest to help with such port? Currently we're underpowered to work on more platforms.

acassis, 2016-01-06:
Hi Gustavo,
Thank you for your reply.
Currently I'm no using Soletta Project, I just opened an issue as suggested by Alsemo:
https://acassis.wordpress.com/2015/06/18/listing-of-projects-devoted-to-iot/#comment-9304

I'm a NuttX contributor (drivers, microcontrollers ports, etc) as you can see in the git log, but I'm not using it for IoT yet. I will take a look at soletta and compare it with other IoT solutions.

barbieri, 2016-01-07:
Ok! I get it.
If you have interest and know the other OS, the basic port should be a matter of hours. Basically you need to hook into build system and implement the main loop, RioT is very similar and you can grep/find as base

See https://github.com/solettaproject/soletta/wiki/Porting

acassis, 2016-01-09:
Thank you Gustavo!

faizansari, 2016-01-11:
bhai log keya hal he koi help chci to apun betha hai