solettaproject/soletta #1262: Encoder nodes

lpereira, 2016-01-11:
Encoders are devices that allow a circuit to know how much something has moved or turned.  Servo motors employs these encoders, for instance, but they're commonly used in robotics as separate sensors where beefier gearboxes are required and commercial servos would be to expensive or difficult to obtain.

By having encoder node types in Soletta, it's possible to implement some localization algorithms and even a software-controlled servo motor.

References:

    * http://www.dynapar.com/Technology/Encoder_Basics/Quadrature_Encoder/
    * https://en.wikipedia.org/wiki/Rotary_encoder
    * https://web.archive.org/web/20150728092206/http://letsmakerobots.com/node/24031

barbieri, 2016-01-11:
can we make a more specific name? Particularly for the title of this issue and the name of the node type... encoder is too open, may confuse people looking at video or audio encoders :-)

lpereira, 2016-01-11:
Sure. Rotary encoders, quadrature encoders, etc.

commit 24a29042 (2016-04-26): flow/robotics: Add quadrature encoder node type

Encoders are devices that allow a circuit to know how much something
has moved or turned.  Servo motors employs these encoders, for
instance, but they're commonly used in robotics as separate sensors
where beefier gearboxes are required and commercial servos would be to
expensive or difficult to obtain.

This adds a quadrature encoder node type, whose output is 0 if the
shaft is not spinning, 1 if it's spinning clockwise, and -1 if
counter-clockwise.

Closes #1262.
closed 2016-04-26 by bdilly