solettaproject/soletta #1263: PID node type

lpereira, 2016-01-11:
PID (Proportional, Integral, Derivative) controllers are commonly used in robotics in order to control devices based on sensor inputs and a set value.  For instance, it is possible to control the temperature of a furnace by carefully controlling how much heat the heating element should produce in order to keep the temperature as close to the desired value as possible.  The implementation is simple (just an expression with numbers as state), although the tuning process isn't that simple.

barbieri, 2016-01-11:
reference URL and if possible something to add to the name (title, node type) so we don't confuse people with process identifier or other PIDs we may have in computing world.

lpereira, 2016-01-11:
Possibly `servo/pid`, or `control-loop/pid`.  Ideally, the three constants (Kp, Ki, and Kd) would be options (that could be overridden by values received from ports, so that online tuning can be implemented). Same thing for the setpoint. A port for the process variable, and another as output.

The Wikipedia page on PID controllers is very explicative, but the implementation is really the formula in the first paragraph: https://en.wikipedia.org/wiki/PID_controller

I have an implementation here: https://github.com/lpereira/Balancing-Robot/blob/master/Firmware/Firmware.pde#L640

lpereira, 2016-01-12:
Autotuning would be nice. Will see how to implement this.

commit 5469b1c7 (2016-04-26): flow/robotics: Implement PID controller

PID (Proportional, Integral, Derivative) controllers are commonly used
in robotics in order to control devices based on sensor inputs and a
set value.  For instance, it is possible to control the temperature of
a furnace by carefully controlling how much heat the heating element
should produce in order to keep the temperature as close to the desired
value as possible.  The implementation is simple (just an expression
with numbers as state), although the tuning process isn't that simple.

Implementation loosely based off of my old implementation in [1].
 [... cut]
closed 2016-04-26 by bdilly