solettaproject/soletta #1269: review all functions taking `void *data` to use the `const` specifier.

barbieri, 2016-01-13:
In many places such as `sol-coap.h` we have functions that take `void *data`, without the `const` specifier, this will require people to add casts if they are giving a `const` object, which is wrong as they may not modify it in their callbacks.

In other cases such as in `sol-vector.h` we are also missing the specifier. In this case we're not talking about callbacks, but nonetheless people may be storing stuff they are not going to modify later, so there is no reason to require them to cast in order to store the pointers.

ceolin, 2016-01-19:
@barbieri please take a look if it is that you had in mind 
https://github.com/solettaproject/soletta/pull/1283

barbieri, 2016-01-19:
looks good

closed 2016-01-20 by ceolin