solettaproject/soletta #1273: locale: Attach monitor receives locale information only once

gabrielschulhof, 2016-01-14:
Steps to reproduce:
  0. Compile https://gist.github.com/gabrielschulhof/12f2d6c47197c7b91d35 from the soletta repo root after having built soletta:
    
    ```BASH
    cc -g -Wall -O0 \
        -I$(pwd)/build/soletta_sysroot/usr/include/soletta \
        -lsoletta -L$(pwd)/build/soletta_sysroot/usr/lib \
        -Wl,-rpath $(pwd)/build/soletta_sysroot/usr/lib \
        monitor-locale.c -o monitor-locale
    ```
  0. Run ```./monitor-locale```
  0. In another terminal, as root, run ```localectl set-locale LANG=de_DE.UTF-8``` or whatever language code is *not* currently set for ```LANG```.
  0. You will observe that ```monitor-locale``` faithfully outputs the new locale.
  0. Repeat the ```localectl set-locale ...``` command from above, but with another language code.
  0. You will observe that ```monitor-locale``` has no additional output in response to the second locale change.

After setting the locale a second time, ```monitor-locale``` *should* output the new locale information again, right?

cabelitos, 2016-01-14:
Note that this problem is happening in the Linux micro backend.
I will take a look, ASAP.

cabelitos, 2016-01-22:
https://github.com/solettaproject/soletta/pull/1307

What you guys think about this fix?

cabelitos, 2016-01-26:
Fix is now merged. closing.
Commit: d662d5856f6ea45f1c2cafc57cb18451307957d6

closed 2016-01-26 by cabelitos