solettaproject/soletta #1318: OIC Client: Platform and server info bindings: The callback must be called even on error

gabrielschulhof, 2016-01-26:
Currently there's an interoperability problem between soletta and iotivity (which I'm investigating): Soletta is unable to parse platform and server information produced by iotivity servers.

In the process, I've noticed that soletta produces a warning that it was unable to parse the payload:
```
WRN: ./src/lib/comms/sol-oic-client.c:324 _platform_info_reply_cb() Could not parse payload
```
and it does not call the callback at all.

Now, it is my assumption that the callbacks associated with the functions ```sol_oic_client_get_platform_info()```, ```sol_oic_client_get_platform_info_by_addr()```, ```sol_oic_client_get_server_info()```, and ```sol_oic_client_get_server_info_by_addr()``` will only be called once, after which you must re-attach them to get them called again.

If my assumption is correct, then that means that soletta must call the callbacks even in the case of failure, because things that were allocated in anticipation of the callback getting called must be freed inside the callback. Otherwise, things will leak. This is the node.js modus operandi for async callbacks, BTW.

In the case of the node bindings, persistent references to the Javascript callback given during the Javascript-equivalent of the above four function calls are allocated during the execution of the bindings, and freed in the callback. Thus, those references will leak.

A possible solution would be calling the callback with NULL for the information structure pointer to indicate an error.

otaviobp, 2016-01-26:
Just to make sure we are in the same page, Soletta is suppose to work with Iotivity-1.0.1. The payload content changed a lot from development versions and the 1.0.1 released and I think it has changed from 1.0.1 to master. Just make sure you are using the correct version. And if you are testing Soletta against an example in Iotivity repository, can you tell me which example you are using?

I used soletta as client to get the server information for examples in iotivity and it is really failing for some examples because iotivity is not setting the device name and this field is set as mandatory in OIC doc. I'll think in a solution for this. (samples/linux/SimpleClientServer/ocserver and samples/linux/SimpleClientServer/ocserverbasicops samples are suppose to work. If you have problems with this samples, let me know).

When getting platform information I also had some mandatory fields missing in iotivity payload and I realized that there are some optional fields that are considered mandatory by soletta. I'll fix that.

Your assumptions are correct. In previous versions, callbacks were never called in timeout or errors. I changed that for in some cases, but it looks that I didn't that for server/platform information getters. I'll work on that.

Thank you for the feedback.

otaviobp, 2016-01-26:
Callback is always being called now, in pr #1325

otaviobp, 2016-01-28:
fixed

closed 2016-01-28 by otaviobp