solettaproject/soletta #1320: OIC Client: Unable to read platform and server information produced by iotivity

gabrielschulhof, 2016-01-26:
#1318 warrants a second issue where the lack of interoperability with ```iotivity``` is the central problem. Basically, calling ```sol_oic_client_get_platform_info()```, ```sol_oic_client_get_platform_info_by_addr()```, ```sol_oic_client_get_server_info()```, and ```sol_oic_client_get_server_info_by_addr()``` on a server powered by ```iotivity``` results in
```
WRN: ./src/lib/comms/sol-oic-client.c:324 _platform_info_reply_cb() Could not parse payload
```
resp.
```
WRN: ./src/lib/comms/sol-oic-client.c:386 _server_info_reply_cb() Could not parse payload
```

otaviobp, 2016-01-26:
I replied about that in #1318. Working on that.

otaviobp, 2016-01-26:
Making some mandatory fields optional in #1325. Note that some iotivity examples are sending empty platform info responses. In those cases callback will be called with info == NULL.

otaviobp, 2016-01-28:
fixed.

closed 2016-01-28 by otaviobp