solettaproject/soletta #1327: check_cflags iterates over each letter, not each option of the .json file

thiagomacieira, 2016-01-26:
Take this test:

```json
    {
      "dependency": "check_cflags",
      "type": "cflags",
      "cflags": [
        "-Wno-cpp",
       "-Wno-#warnings"
      ],
      "comment": "append to CFLAGS_NO_SOURCE_WARNINGS",
      "append_to": "CFLAGS_NO_SOURCE_WARNINGS"
    },
```

config.log shows:

```
Command: gcc -Werror -Wno-cpp -Wno-#warnings /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp4w3xl9af.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp4w3xl9af.c-bin 
exit code: 1
Command output(stderr):
error: unknown warning option '-Wno-cpp' [-Werror,-Wunknown-warning-option]

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  - /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpw9fqd2ie.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpw9fqd2ie.c-bin 
exit code: 1
Command output(stderr):
clang: error: -E or -x required when input is from standard input

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  W /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpboayr9fu.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpboayr9fu.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'W'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  n /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpcqptonzx.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpcqptonzx.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'n'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpe2x8dkam.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpe2x8dkam.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'o'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  - /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp7eyk2rke.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp7eyk2rke.c-bin 
exit code: 1
Command output(stderr):
clang: error: -E or -x required when input is from standard input

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  c /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpw3lfiakk.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpw3lfiakk.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'c'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  p /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp03dgbpma.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmp03dgbpma.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'p'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
Command: gcc -Werror  p /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpvrbjqled.c -o /var/folders/_q/byf907ss6bb130qvj9fqk2nr0000gq/T/tmpvrbjqled.c-bin 
exit code: 1
Command output(stderr):
clang: error: no such file or directory: 'p'

Compiling source code:

int main(int argc, char **argv){
  return 0;
}
```

Note how it tests each letter after "-Werror"

barbieri, 2016-01-27:
ok, I looked `dependency-resolver.py` and it is wrong handling failure. I'll fix and send a patch

barbieri, 2016-01-27:
See #1344

thiagomacieira, 2016-01-27:
Fixed by 93210b176e6bd62e8d76748cabdbb2a546c5eac2.

closed 2016-01-27 by thiagomacieira