solettaproject/soletta #1328: Double slash in some Make rules break compilation

thiagomacieira, 2016-01-26:
For example:

```
make: *** No rule to make target `src//modules/flow/oic/oic.json', needed by  `build/soletta_sysroot/usr/local/Soletta/include/soletta/sol-flow/oic.h'.  Stop.
```

Note the double slash in "src//modules". There is a rule to generate this file, but with a single slash.

I don't know which Makefile contains those rules. Grep can't find anything.

bdilly, 2016-01-27:
@ceolin maybe @dorileo can help you with that.

ceolin, 2016-01-27:
i can help with this. Thiago how can i reproduce it ? I've never faced this problem in my system.

thiagomacieira, 2016-01-27:
I don't know what caused it. I got the latest version and typed gmake.

On OS X.

ceolin, 2016-01-27:
ok, i'll check it correctly

ceolin, 2016-01-27:
fixed https://github.com/solettaproject/soletta/commit/9f60a08b67bbbc85b2d49c93f5af68bded453fd7

closed 2016-01-27 by ceolin