solettaproject/soletta #1331: Platform pin multiplexer is failing to load when sol_init() is called from a module

nagineni, 2016-01-26:
Platform pin multiplexer is failing to load on target device, when sol_* APIs are called from a module(e.g nodejs module) that was loaded with dlopen by an executable(e.g node) using RTLD_LOCAL flag.

Here is the log:
`INF:sol-pin-mux ./src/lib/common/sol-pin-mux.c:82 _load_mux() Could not load platform pin multiplexer '//usr/lib/soletta/modules//pin-mux//intel-edison-rev-c.so': //usr/lib/soletta/modules//pin-mux//intel-edison-rev-c.so: undefined symbol: sol_log_global_domain`

Note that the issue is not reproducible when we open an executable that linked with libsoletta directly. I already checked this by compiling a simple example that uses the same sol C API, and running it on Edison.

bdilly, 2016-01-26:
@mbelluzzo could you please take a look on this issue? thanks

mbelluzzo, 2016-01-27:
@nagineni, just to help me to find out what is goind on: what happens if you disable the pinmux during configuration?

barbieri, 2016-01-27:
This fails for sure, see the C snippet below that can be tested as:

```sh
SOL_LOG_LEVEL=4 SOL_BOARD_NAME=intel-edison-rev-c  /tmp/t libsoletta.so
```

Running without `SOL_BOARD_NAME` in my desktop it works, while using edison board name it will fail with the exact message.

```c
#include <dlfcn.h>
#include <stdio.h>
#include <stdlib.h>

int main(int argc, char *argv[])
{
    void *h;
    int (*f_init)(void);
    void (*f_shutdown)(void);
    int retcode = EXIT_FAILURE;
    int r;

    h = dlopen(argv[1], RTLD_LOCAL | RTLD_NOW);
    if (!h) {
        fprintf(stderr, "error: %s: %s\n", argv[1], dlerror());
        return 1;
    }

    f_init = dlsym(h, "sol_init");
    if (!f_init) {
        fprintf(stderr, "error: no sol_init() in %s: %s\n", argv[1], dlerror());
        goto end;
    }

    f_shutdown = dlsym(h, "sol_shutdown");
    if (!f_shutdown) {
        fprintf(stderr, "error: no sol_shutdown() in %s: %s\n", argv[1], dlerror());
        goto end;
    }

    r = f_init();
    if (r < 0) {
        fprintf(stderr, "sol_init() = %d\n", r);
        goto end;
    }

    f_shutdown();

end:
    dlclose(h);
    return retcode;
}
```

barbieri, 2016-01-27:
there are likely more than one bug. I found one already: intel-common.c does declare its own log-level, but doesn't use it. It misses:
```c
#define SOL_LOG_DOMAIN &_intel_mux_log_domain
...
#include "sol-log-internal.h"
...
SOL_LOG_INTERNAL_DECLARE(_intel_mux_log_domain, "intel-mux");
```

But the global log domain should be usable. Then it's likely missing linkage.

barbieri, 2016-01-27:
indeed none of the modules link to libsoletta, it happened to work as the main binary was linked and thus 'leaking' the symbols.

barbieri, 2016-01-27:
Please check #1348, it should fix your issue

nagineni, 2016-01-28:
@barbieri @bdilly @mbelluzzo, thanks a lot for looking into the issue and providing a fix quickly. 

@barbieri you are right. pin mux module failure is just one case. I can see that other dlopen calls are also failing to load the modules(like linux-micro-efi-update.so) due to missing symbols when libsoletta loaded with dlopen.

I will verify the fix on target device now. I'm sure it will fix the issue :)

nagineni, 2016-01-28:
PR #1348 solves this issue. I am able to blink the LED on edison from nodejs bindings.

mbelluzzo, 2016-01-28:
\o/

commit 206f6c23 (2016-01-28): Soletta modules should link to libsoletta.

If we do not link to libsoletta, then whenever a module (ie: pin-mux,
flow, linux-micro) are loaded from a binary that is NOT linked with
libsoletta, then it will fail to find soletta's symbols.

It didn't show before because all of our tests and binaries are linked
with it. But using it as a node-js module or a simple C application
that calls `dlopen("libsoletta.so")` and calls `sol_init()` may cause
errors.

This should fix #1331.
closed 2016-01-28 by mbelluzzo