solettaproject/soletta #1371: humidity sensor reading result into segment fault.

guchaojie, 2016-01-29:
./humidity.fbp
WRN: ./src/lib/io/sol-iio.c:697 sol_iio_open() Could not disable buffer for device1
WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:1003 humidity_create_device_cb() mdata->device == NULL
Segmentation fault

dmesg -c
[1128737.017182] sol-fbp-runner[8534]: segfault at 656d69b0 ip b7680739 sp bf9120d0 error ffff0004 in libsoletta.so[b7600000+fd000]

edersondisouza, 2016-01-29:
> ./humidity.fbp
> WRN: ./src/lib/io/sol-iio.c:697 sol_iio_open() Could not disable buffer
> for device1

This first warning should have more warnings before, please send them 
too - it would be interesting also run using strace, like:

   strace -ff sol-fbp-runner humidity.fbp

So we can get the real issue on this.

> WRN:sol-flow-iio ./src/modules/flow/iio/nodes.c:1003
> humidity_create_device_cb() mdata->device == NULL
> Segmentation fault

This is really weird. The device creation failed, as shown in the 
previous warning. Anyway, crashes are not ok. Could you please run this 
code with valgrind, so we get where this segfault happens? Don't forget 
to build Soletta for debug. I think there's a Kconfig flag, but you can 
simply use
"make CFLAGS='-g'"

Cheers,

guchaojie, 2016-02-03:
@edersondisouza PR#1387 has been submitted to fix this bug.

edersondisouza, 2016-02-17:
As #1424 solved this issue - acked by @guchaojie - I'll close it.

closed 2016-02-17 by edersondisouza